ProteoWizard/pwiz #29: msconvert 'loadlocale.c:129'

jspmccain, 2018-04-20:
Hello,

I'm working on an Ubuntu computer and trying to access msconvert via the terminal. I'm getting an error message that I'm having a tough time interpreting:

msconvert: loadlocale.c:129: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
Aborted (core dumped)

Other command line functions in ProteoWizard, like msaccess, do not give the same error and work fine. 

I'm running:

Distributor ID: Ubuntu
Description:    Ubuntu Bionic Beaver (development branch)
Release:        18.04

Any thoughts are appreciated!

chambm, 2018-04-20:
Does it work if you set `export LC_ALL=C`?

jspmccain, 2018-04-20:
Yes that works perfect, thanks! Now just to figure out converting raw files to mzML on Linux...

closed 2018-04-20 by jspmccain

eladnoor, 2019-10-29:
Hello,

I just encountered the same issue in release-3_0_19302_c1e9fb6 (I'm using Ubuntu 19.10).
Perhaps it will be helpful to fix it or at least post this solution on the [download](http://proteowizard.sourceforge.net/download.html) page?