ProteoWizard/pwiz #41: Confusing Requirements for BiblioSpec Compilation

DarioS, 2018-04-30:
The [Download and Build](https://skyline.ms/wiki/home/software/BiblioSpec/page.view?name=Download%20and%20build) webpage states "You will need Visual Studio 2013 to build." However, this doesn't seem applicable to installing it when the operating system is Linux. Could it be changed? There is also no branch named "trunk", so I downloaded skyline_4_1 and ran quickbuild.sh which ended with:

```
Building pwiz...
Jamroot.jam:446: in modules.load
*** argument error
* rule constant ( name : value + )
* called with: ( PWIZ_GIT_BRANCH :  )
* missing argument value
```

chambm, 2018-04-30:
Sorry for the confusion. Did you clone with git or downloaded it some other way? If so, try the master branch (it's the git equivalent of trunk).

DarioS, 2018-05-01:
I downloaded master and I get 

```
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
```

I'm not sure why it has to be a git repository. The instructions state "Download (or clone with git) the trunk/pwiz directory from the ProteoWizard git repository." So, it seems that `git clone` is optional. I clicked the Download ZIP button and used `unzip` to decompress it which seems to be acceptable, according to the instructions. Also, should it be necessary for users to download all 631 MB of ProteoWizard just to compile the BiblioSpec spectral library file format converter? It doesn't seem modular. Are Ubuntu DEBs of it available from somewhere?

chambm, 2018-05-01:
We recently switched from SVN to Git and there have been growing pains. The need to get the branch and revision info via the `git` command rather than through SVN properties is one of them. The BiblioSpec build documentation has only been lightly updated to reflect this transition. Here's a diff that will make the git commands optional (I will get this integrated into master soon):
https://gist.github.com/chambm/bc6e370dd490ed491723a56039969fd0

Here's a subset source tarball which should only contain the files necessary to build BiblioSpec:
http://teamcity.labkey.org/viewLog.html?buildId=576515&tab=artifacts&buildTypeId=BiblioSpecSubsetSourceTarball&guest=1
The `without-t` version omits test files to save space.

chambm, 2018-05-02:
@DarioS let me know if this works for you so I can close the issue. Thanks.

DarioS, 2018-05-03:
I downloaded bibliospec-src-without-t-3_0.tar.bz2 from the TeamCity website, extracted it and ran `quickbuild.sh` but the error remains the same:

> fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

I was planning to wait a few days and download ProteoWizard again once the Gist was incorporated into the repository.