ProteoWizard/pwiz #124: startTimeStamp in Waters RAW file conversion

adwolfer, 2018-07-05:
Hi,

When using MSConvert on Waters .raw files to mzML, the `startTimeStamp` field is populated using the _HEADER.TXT_ acquisition date and time fields.
However it appears that if the acquired time contains '_08_' or '_09_' as the hour, minute or second; the value is replaced by '_00_' in the `startTimeStamp`. For example:

`$$ Acquired Date: 05-Sep-2017`
`$$ Acquired Time: 08:09:08`
Results in:
`startTimeStamp="2017-09-05T00:00:00Z"` instead of `startTimeStamp="2017-09-05T08:09:08Z"`

We experienced this issue with multiple datasets and it can be simulated by simply changing the acquisition time in _HEADER.TXT_ in any raw data. (tested in today's build _ProteoWizard 3.0.18182.5406cdbf0_)

Thanks and apologies for cross-posting this in the support-mailing list

Best regards

stanstrup, 2018-08-19:
I can confirm that this is an issue. Highly problematic when I use the startTimeStamp to do intensity drift correction.

chambm, 2018-08-30:
Fixed with Boost upgrade in https://github.com/ProteoWizard/pwiz/pull/82 - so official downloads will have this fixed.

stanstrup, 2019-02-06:
Is this supposed to have been fixed in the compiled version?
I am still seeing this issue unfortunately with:
```
ProteoWizard release: 3.0.19014 (f9d5b8a3b)
Build date: Jan 14 2019 20:20:10
```

If not, will we see the fix available soon?

chambm, 2019-02-06:
Interesting. I'm not sure why I blamed this on Boost before but digging into it this time revealed that it's because of my overriding `lexical_cast` with `strtol` calls in `optimized_lexical_cast.hpp`. The time parser was looking at `09` and treating it like an octal (base 8) number. I'm updating the code to always treat such conversions as base 10, which appears to be what basic lexical_cast (and C++ istream) does.

Fixed in #439

closed 2019-02-06 by chambm