ProteoWizard/pwiz #144: Filtering ITMS/FTMS from Lumos data files losing all scan data

chrashwood, 2018-07-27:
Dear Proteowizard developers,

I hope you are well. I'm currently having some trouble with msconvert regarding filtering of a Thermo Lumos raw data file containing multiple mass analyzer steps. I am using ProteoWizard 3.0.8725 x64. Direct and non-filtered conversion from raw to mzML, then subsequent import into Skyline works fine and the filter string names are preserved in the unfiltered mzML file. 

Curiously however, if implementing filtering for only FTMS (--filter "analyzer FT" OR --filter "analyzer orbi" OR "--filter "analyzerType FTMS", all scans are lost with only the chromatogram/metadata preserved. I have provided a copy of the command line process run below:

`W:\1_RAW_Data\4_Lumos>>
msconvert Lumos_data.raw --filter "analyzer FT"
format: mzML
    m/z: Compression-None, 64-bit
    intensity: Compression-None, 32-bit
    rt: Compression-None, 64-bit
ByteOrder_LittleEndian
 indexed="true"
outputPath: .
extension: .mzML
contactFilename:

filters:
  analyzer FT

filenames:
  Lumos_data.raw

processing file: Lumos_data.raw
writing output file: .\Lumos_data.mzML
`

This same process works fine with my Velos Orbitrap data so I don't believe my commands to be the issue.

Cheers,
Chris

chambm, 2018-07-30:
What are your <instrumentConfiguration> sections in the resulting mzML?

chrashwood, 2018-07-31:
> What are your sections in the resulting mzML?

Thanks for the quick reply.

Comparing the filtered vs non-filtered mzml conversion:
run id spectrumList is missing
run id chromatogramList is the same.

index name="spectrum" is different (no details present for this section and
no offset values)
index name="chromatogram" is different (TIC intensity is 0.001% of the
non-filtered mzML file)

indexedmzML, mzML, cvList,
fileDescription, referenceableParamGroupList, softwareList,
instrumentConfigurationList
and dataProcessingList are the same.

chambm, 2018-07-31:
I mean what are the contents in the `<instrumentConfigurationList>`

chrashwood, 2018-07-31:
Sorry for the misinterpretation:

 <instrumentConfiguration id="IC1">
        <referenceableParamGroupRef ref="CommonInstrumentParams"/>
        <componentList count="4">
          <source order="1">
            <cvParam cvRef="MS" accession="MS:1000398"
name="nanoelectrospray" value=""/>
            <cvParam cvRef="MS" accession="MS:1000485" name="nanospray
inlet" value=""/>
          </source>
          <analyzer order="2">
            <cvParam cvRef="MS" accession="MS:1000081" name="quadrupole"
value=""/>
          </analyzer>
          <analyzer order="3">
            <cvParam cvRef="MS" accession="MS:1000484" name="orbitrap"
value=""/>
          </analyzer>
          <detector order="4">
            <cvParam cvRef="MS" accession="MS:1000624" name="inductive
detector" value=""/>
          </detector>
        </componentList>
        <softwareRef ref="Xcalibur"/>
      </instrumentConfiguration>
      <instrumentConfiguration id="IC2">
        <referenceableParamGroupRef ref="CommonInstrumentParams"/>
        <componentList count="4">
          <source order="1">
            <cvParam cvRef="MS" accession="MS:1000398"
name="nanoelectrospray" value=""/>
            <cvParam cvRef="MS" accession="MS:1000485" name="nanospray
inlet" value=""/>
          </source>
          <analyzer order="2">
            <cvParam cvRef="MS" accession="MS:1000081" name="quadrupole"
value=""/>
          </analyzer>
          <analyzer order="3">
            <cvParam cvRef="MS" accession="MS:1000083" name="radial
ejection linear ion trap" value=""/>
          </analyzer>
          <detector order="4">
            <cvParam cvRef="MS" accession="MS:1000253" name="electron
multiplier" value=""/>
          </detector>
        </componentList>
        <softwareRef ref="Xcalibur"/>
      </instrumentConfiguration>
    </instrumentConfigurationList>

chambm, 2018-07-31:
Hmm, are you sure that your data contains FT scans? The scan filters say FTMS? Can you send a RAW file?

chrashwood, 2018-07-31:
100% sure. It's a triggered neutral loss experiment. I've compressed the raw file so it's a 400MB zip archive. Do you have a preferred method to share the data? It's not a sensitive method/sample so can be public.

chambm, 2018-07-31:
This will work for non-sensitive data: https://skyline.ms/filecontent/home/support/file%20sharing/begin.view?

chrashwood, 2018-07-31:
Thanks. It's uploaded and available now.

chambm, 2018-08-30:
I can reproduce the problem and have figured out the cause, I just haven't gotten to fix it yet. (the filter only checks the first analyzer, when it really should check the last, or all)

chrashwood, 2018-08-30:
That's great to hear, thanks Matt! Is the single mass analyzer assumption specific only for msconvert or are other programs part of the Proteowizard suite such as msaccess subject to this limitation?

chambm, 2018-09-05:
Should be fixed in #204.

closed 2018-09-06 by chambm

jspaezp, 2021-01-24:
Sorry for resurrecting this issue but I am having the same problem using the `msconvert` in `ProteoWizard 3.0.18215.f6864290f`

I am using this file from PRIDE
Pride repo: PXD004732
Test File:  01625b_GB2-TUM_first_pool_10_01_01-2xIT_2xHCD-1h-R1.raw

```bash
MSCONVERT="/c/Program Files/ProteoWizard/ProteoWizard 3.0.18215.f6864290f/msconvert.exe"
TESTFILE="01625b_GB2-TUM_first_pool_10_01_01-2xIT_2xHCD-1h-R1.raw"
count_filters() {
    grep -P "filter" ./out/*.mzML | grep -oP "value=\".*?ms(\d|\s)" | sort | uniq -c 
}

"${MSCONVERT}" ${TESTFILE} --filter "peakPicking true 1-" -o out/
count_filters
#   28830 value="FTMS + c NSI d Full ms2
#    3227 value="FTMS + p NSI Full ms
#   28830 value="ITMS + c NSI r d Full ms2

"${MSCONVERT}" ${TESTFILE} --filter "analyzer orbi" --filter "peakPicking true 1-" -o out/
count_filters
# EMPTY

"${MSCONVERT}" ${TESTFILE} --filter  "peakPicking true 1-" --filter "analyzer orbi" -o out/
count_filters
# EMPTY

"${MSCONVERT}" ${TESTFILE} --filter "analyzer orbi" -o out/
count_filters
# EMPTY

"${MSCONVERT}" ${TESTFILE} --filter "analyzer FT" -o out/
count_filters
# EMPTY

"${MSCONVERT}" ${TESTFILE} --filter "analyzer FTMS" -o out/
count_filters
# EMPTY
```

Is there anything I am doing wrong?

Kindest wishes,
Sebastian

chambm, 2021-01-24:
Try a newer version of pwiz. Even though this issue was fixed in 2018, it seems your version manages to predate it.

jspaezp, 2021-02-20:
Noticed the error was due to the path not using `msconvert` from the actual updated installation (so I guess the issue is that updating pwiz does not remove older versions, which in itself could be a design choice...)

Thanks for the reply.