ProteoWizard/pwiz #155: Cannot Find MS3 SPS data

trestoh, 2018-08-15:
ProteoWizard Devs,

I've been trying to find a way to extract SPS data from mzML files made with msconvert but so far am not having any luck. My raw files are from a Thermo Fusion Lumos, and looking through the mzML files by hand I can't find any tag or content that looks like SPS data. I know the data is in the raw somewhere because I can see it when I open the files in Thermo's software, but they seem to be getting lost in the conversion. My only guess so far is that this might be related to the filter strings. I've seen examples online where the sps designation seems to be included in the filter string, but in both my mzML and when opened in Thermo's software my filter strings look like the following: "FTMS + c NSI d Full ms3 608.8657@cid30.00 430.7821@hcd65.00 [100.0000-500.0000]".

Is there some setting I'm missing maybe? I apologize if this is not a software issue and is just a user error. 

Regards,
Trent

pwilmart, 2018-08-26:
Hi Trent,
You could try RawQuant (https://github.com/kevinkovalchik/RawQuant.git) to get some useful quant tables. 

SPS-MS3 data is complicated. In between the bracketing MS1 scans, you have MS2 and MS3 scans. They are in sequential order (the MS3 always happens after its MS2 scan), but the pattern is variable (max speed algorithms and the complicated parallel acquisition modes have some implications). You can have a few MS2 scans, then you start seeing the MS3 scans sometimes in a row or mixed with more MS2 scans. The MS3 scans are relatively slow so the ion trap can do multiple MS2 while waiting. 

You have to track the scan numbers and m/z values (appropriately rounded to the right number of decimal places) of the MS2 scans (i.e. in something like a Python dictionary). Then, when an MS3 is encountered, you have to grab the scan number and extract the MS2 m/z out of the filter string. Your example above has the MS2 ion at 608.8657 m/z. Then you can figure out what scan number for the MS2 corresponds to that m/z value and get the mapping between MS2 scan number and MS3 scan number. You also have to get the reporter ion data from all of the MS3 scans and then store the quantitative data and scan mappings in a data structure for later lookup. After you have done a database search, filtered out incorrect PSMs, mapped PSMs to proteins, applied parsimony logic, and determined which peptides are usable for quantification (typically unique peptides), then you can get the lists of all PSMs for those peptides, fetch the quant data for the reporter ions from your data structure and summarize it all in some appropriate way (total protein intensity, some median protein ratio, etc.).
Cheers,
Phil

chambm, 2018-08-30:
Do the ms3 filter lines show up in the mzML? I don't see the sps tag in that filter.

trestoh, 2018-08-30:
I should have been more clear, I'm specifically looking for which fragments were selected to perform the sps, which in Thermo's software looks like "SPS Masses: 406.37, 425.67, 444.21, 467.01, 487.96, 542.31, 603.45, 621.86," and is listed as part of the scan's header. Matt: I'm not sure exactly what you mean by MS3 filter lines, but there does not appear to be anything "extra" in the MS3 scans in the mzML that isn't also in the MS2s. That is how all the MS3 filters look, no sps tag, and I'm wondering if that might be why I don't have the SPS Masses fields, even though I know they must be in the raws.

chambm, 2018-08-30:
I've never seen that field in the scan header. I've only known of sps from the "sps" tag in filter line. Which your files don't have (or maybe it's on the MS2 filter lines instead?). But obviously if it's in your scan header that's an alternative place we need to look. However we don't actually do anything with the "SPS on" flag.

If the "SPS Masses" header is in the MS2, and its purpose is to describe which MS3s were triggered, then as far as I can tell, it's superfluous information. You will find that information stored in the MS3s' precursor tags. And if it's working as intended, the MS3 scans' precursors will also point at the MS2 scan that triggered it (the spectrumRef attribute).

So if you wanted to group together the MS3s corresponding to each MS2, you just need to map from the MS2 spectrumRef to the spectrum ids of the MS2. ProteoWizard takes care of determining which MS3 was triggered by which MS2 (it makes sure the MS3's MS2 precursor matches the MS2).

pwilmart, 2018-08-30:
RawQuant gives you the lists of MS2 fragment ions corresponding to the notches. That is all layered on top of MSFileReader so it must be in the RAW file somewhere.

chambm, 2018-08-31:
Precursor scan ids are not directly accessible in MSFileReader. They might be in RawFileReader (MSFR's 64-bit-only replacement), but I haven't tried finding it because my own code works fine for identifying precursor scans.

closed 2018-09-05 by chambm