ProteoWizard/pwiz #187: Support for Thermo TSQ Altis QQQ

FarmGeek4Life, 2018-08-30:
MSConvert Version (x64): 3.0.18230.1e0a1ad53

Trying to convert a RAW file produced by a Thermo TSQ Altis QQQ I get the error "[Reader_Thermo::fillInMetadata] unable to parse instrument model; please report this error to the ProteoWizard developers with this information: model(TSQ Altis) name(TSQ Altis); if want to convert the file anyway, use the ignoreUnknownInstrumentError flag".

metadata.txt is the sanitized output from ThermoRawMetaDump.exe. This is SRM data, but in separate processing I have learned that the instrument method is for some reason unreadable by MSFileReader 3.0, but can be read by MSFileReader 3.1 (which is 64-bit only) and also by RawFileReader.

[metadata.txt](https://github.com/ProteoWizard/pwiz/files/2337516/metadata.txt)

chambm, 2018-09-05:
Added in #204 (but it doesn't have its own CV param yet; it'll show up as just "TSQ", which I thought was at least less misleading than "TSQ Quantum" or some other newer TSQ).

FarmGeek4Life, 2018-09-05:
there is the CV term MS:1002874, but that doesn't exist in the pwiz CV enum yet...

chambm, 2018-09-06:
Now that's fixed too. Thanks for checking the CV.

closed 2018-09-06 by chambm