ProteoWizard/pwiz #188: Quameter error when first spectrum is not MS1

alchemistmatt, 2018-08-30:
We run Quameter in automation and it generally works well.  However, we occasionally get a Thermo .raw file where the first spectrum (or spectra) is MS2, then later an MS1 spectrum appears, followed by more MS2 spectra. For data files that start with an MS2 spectrum, Quameter reports
`Error processing ID-free metrics: error reading spectrum index 0 (No MS1 spectrum found before controllerType=0 controllerNumber=1 scan=1)`

I found a workaround -- update the default filter to include a scan range:
` -SpectrumListFilters: "peakPicking true 1-;threshold absolute 0.00000000001 most-intense; scanNumber [14,1000000]"`

However, it would be nice if Quameter could automatically skip the initial MS2 spectra and resume normal operation once an MS1 spectrum is found.  If the entire file is MS2 spectra, it should report an error.  Is this something that could be added?

chambm, 2018-08-31:
It's been a while since I've worked on that code. Is it a DDA file that has an MS2 before the first MS1?

alchemistmatt, 2018-08-31:
Yes, that's right, DDA, but the first scan is MS2.  Scan 14 is MS1.  My colleague explained to me that he configured the instrument to do that on purpose, due to some quirkiness with the Xcalibur data acquisition software.

chambm, 2018-09-04:
See if it's fixed here:
http://teamcity.labkey.org/repository/download/Bumbershoot_Windows_X86_64/625247:id/bumbershoot-bin-windows-vc141-release-3_0.tar.bz2?guest=1

alchemistmatt, 2018-09-04:
Yes, 9c733776e13bf7365f937d3dde5efe0dc5a22804 fixed the issue; I successfully processed the dataset that starts with several MS2 spectra.  Great job!

However, now there is a new bug.  It's minor, but possibly an easy fix.  
The metric for `MS1-TIC-Q2` reports `-nan(ind)`

When I instruct quameter to start processing at the first MS1 scan, that metric reports 0.613.  Hmm, examining the metrics in greater detail I see that all of the TIC-based metrics have changed (vs. a January 2016 build of Quameter), some quite a bit.  And, the MS1-Density metrics are 5 fold higher.  This is possibly related to using ThermoFisher.CommonCore.RawFileReader.dll instead of MSFileReader.XRawfile2.dll

I analyzed a separate dataset (one that does start with a MS1 scan) and the metrics agree much better.  However, the MS1-Density metrics are still 5 fold higher.  I have attached two tab-delimited text files comparing old vs. new values.  Up to you whether you want to go down the rabbit hold of tracking down the source of the changes.

[Quameter_Results_Comparison_ABP_B4.txt](https://github.com/ProteoWizard/pwiz/files/2349896/Quameter_Results_Comparison_ABP_B4.txt)
[Quameter_Results_Comparison_BTK_TFEdigest.txt](https://github.com/ProteoWizard/pwiz/files/2349897/Quameter_Results_Comparison_BTK_TFEdigest.txt)

chambm, 2018-09-04:
Did you use the same peak picking settings between the two? Density is just distribution of number of data points. I can't imagine that changing too much between MSFR and RFR, either for profile or centroid spectra.

I think I see what's causing the nan issue. I should skip those leading MS2s instead of letting them participate in the metrics.

alchemistmatt, 2018-09-04:
Yes, identical settings
```
ChromatogramMzLowerOffset: "10ppm"
ChromatogramMzUpperOffset: "10ppm"
       ChromatogramOutput: "0"
               Instrument: "orbi"
              MetricsType: "idfree"
 MonoisotopeAdjustmentSet: "[0,0] "
          NumChargeStates: "3"
           OutputFilepath: "Quameter_IDFree_QC_Shew_New.tsv"
            RawDataFormat: "RAW"
              RawDataPath: ""
              ScoreCutoff: "0.050000000000000003"
      SpectrumListFilters: "peakPicking true 1-;threshold absolute 0.00000000001 most-intense"
```
I just analyzed a 3rd dataset to compare the Quameter scores.   Again, the scores are mostly the same, but the MS1-density ones changed drastically.

Score | Old | New
-- | -- | --
MS1-Density-Q1 | 403 | 2743
MS1-Density-Q2 | 477 | 3231
MS1-Density-Q3 | 548 | 3790

alchemistmatt, 2018-09-04:
Interestingly, MS2-Density is identical

Score | Old | new
-- | -- | --
MS2-Density-Q1 | 135.25 | 135.25
MS2-Density-Q2 | 268 | 268
MS2-Density-Q3 | 426 | 426

It's almost like the MS1 spectra are not getting centroided, despite `peakPicking true 1-`

chambm, 2018-09-04:
Hmm. Well, do the new MS1-Density numbers change when you set it to `peakPicking true 2-`?

alchemistmatt, 2018-09-04:
No change with `-SpectrumListFilters "peakPicking true 2-;threshold absolute 0.00000000001 most-intense"`.  The only way to make them change is to use `-SpectrumListFilters "" which disables centroiding and causes the MS1-Density values to double.

Filename | Old | peakPicking true 1-; threshold absolute   0.00000000001 most-intense | peakPicking true 2-; threshold absolute   0.00000000001 most-intense | threshold absolute 0.00000000001 most-intense | SpectrumListFilters=""
-- | -- | -- | -- | -- | --
MS1-Density-Q1 | 403 | 2743 | 2743 | 2743 | 5568
MS1-Density-Q2 | 477 | 3231 | 3231 | 3231 | 6457
MS1-Density-Q3 | 548 | 3790 | 3790 | 3790 | 7416
MS2-Density-Q1 | 135.25 | 135.25 | 135.25 | 135.25 | 135.25
MS2-Density-Q2 | 268 | 268 | 268 | 268 | 268
MS2-Density-Q3 | 426 | 426 | 426 | 426 | 426

Next, I'll look for the most recent version of Quameter that used MSFileReader.XRawfile2 and give it a try.

chambm, 2018-09-04:
Note that the 32-bit build still uses MSFR.

alchemistmatt, 2018-09-04:
In an attempt to help track down the issue, I tried a range of Bumbershoot_Windows_X86_64 builds.  I found that [Build 913](http://teamcity.labkey.org/viewLog.html?buildId=621978&tab=artifacts&buildTypeId=Bumbershoot_Windows_X86_64) (2018 Aug 28) produces the expected results but [Build 962](http://teamcity.labkey.org/viewLog.html?buildId=624938&tab=artifacts&buildTypeId=Bumbershoot_Windows_X86_64) (2018 Sep 01) has the 5-fold higher MS1-Density values.

Changes from August 28 to September 1 include the reader switch on August 29 
(`switched to use RawFileReader for reading Thermo RAW files (#170)`)

I'm not in a pressing need for a new version of Quameter; just trying to help you weed out bugs.

FarmGeek4Life, 2018-09-04:
Does [raw_->IsCentroidScanFromScanNumber(scanNumber)](https://github.com/ProteoWizard/pwiz/blob/master/pwiz_aux/msrc/utility/vendor_api/thermo/RawFile.cpp#L859) return false for profile mode FTMS data?

alchemistmatt, 2018-09-04:
Yes (just tested), it returns false for a profile scan
```
Scan 5372, FTMS + p NSI Full ms [400.00-2000.00]; IsCentroidScanFromScanNumber: False
Scan 5373, FTMS + c NSI d Full ms2 603.33@hcd32.00 [100.00-1220.00]; IsCentroidScanFromScanNumber: True
Scan 5374, FTMS + c NSI d Full ms2 681.04@hcd32.00 [100.00-2000.00]; IsCentroidScanFromScanNumber: True
```
and thus `if (centroidResult && raw_->IsCentroidScanFromScanNumber(scanNumber))` is returning false, thereby skipping the call to `GetCentroidStream`

chambm, 2018-09-05:
Ugh. That would do it. I mistakenly thought there was a unit test covering the peak picking. Maybe I meant to do OR. If it's a centroid scan, it presumably wouldn't have profile data, so it wouldn't make sense to call `GetSegmentedScanFromScanNumber` (but obviously calling it with a centroid-only scan does not, in fact, cause an error, or else NOT peak picking a file that has centroid-only scans would be an error).

FarmGeek4Life, 2018-09-05:
So, there is one reason for that check: `GetCentroidStream` doesn't work for profile ion trap data. It took some looking, but I think I figured out the proper way to do it (for profile ion trap data): `Scan::ToCentroid(Scan::FromFile(raw_, scanNumber))` will get a centroid `Scan` object, then getting the `PreferredMasses` and `PreferredIntensities` will give you the needed arrays (or you can get `SegmentedScan`). The gateway check might involve `raw_->GetFilterForScanNumber(scanNumber)->MassAnalyzer != MassAnalyzerType::MassAnalyzerFTMS`. Also, calling `Scan::ToCentroid()` on non-profile data returns the supplied `Scan` object.

FarmGeek4Life, 2018-09-05:
So, after further tests, I think I have an implementation that should work ProteoWizard, with results matching MSFileReader expectations (pardon any incorrectness on the C++, I'm translating from C#):
[Replacing the `if`, but not the `else`](https://github.com/ProteoWizard/pwiz/blob/master/pwiz_aux/msrc/utility/vendor_api/thermo/RawFile.cpp#L859)
```cpp
if (centroidResult && !(raw_->IsCentroidScanFromScanNumber(scanNumber)))
{
    auto centroidStream = raw_->GetCentroidStream(scanNumber, false);
    // With the file I tested, the Masses array was null
    if (centroidStream == nullptr || centroidStream->Masses == nullptr || centroidStream->Masses->Length <= 0)
    {
        auto centroidedScan = Scan::ToCentroid(Scan::FromFile(_raw, scanNumber))->SegmentedScan;
        ToStdVector(centroidedScan->Positions, result->mzArray);
        ToStdVector(centroidedScan->Intensities, result->intensityArray);
    }
    else
    {
        ToStdVector(centroidStream->Masses, result->mzArray);
        ToStdVector(centroidStream->Intensities, result->intensityArray);
    }
}
```

FarmGeek4Life, 2018-09-05:
I do think the mistake in the original mainly consisted of the `!(raw_->IsCentroidScanFromScanNumber(scanNumber))`, because (also tested here) if the scan is centroid mode, then centroidStream will be also be empty; the correct way to get the centroid peaks from a centroid-mode scan is with `GetSegmentedScanFromScanNumber(scanNumber, null)`

chambm, 2018-09-05:
Thanks for the testing @alchemistmatt and @FarmGeek4Life . I've fixed the peak picking and several other issues in #204. I added an option to test peak picking in our vendor testing harness, and enabled that for the Thermo reader. Our test file is from an LTQ Velos and it has both centroid and profile scans so it was well-suited for making sure that code works.

FarmGeek4Life, 2018-09-05:
Note that if someone requests centroided data from a scan that is centroided FTMS, [this code](https://github.com/ProteoWizard/pwiz/pull/204/commits/92c0cf6c75ba89e2aed770ed903720df125b937b#diff-44f51aeb7592c5e2b5911b092755f623R863) will lead to an exception.

chambm, 2018-09-06:
> Note that if someone requests centroided data from a scan that is centroided FTMS, this code will likely lead to an exception
I tested it on centroided orbitrap data and it works. The CentroidStream doesn't actually centroid, it's just accessing the centroided data that's already there. For profile mode FTMS, the data has both the centroids and the profile data. For centroid mode FTMS, the profile data was discarded and only the centroids kept.

FarmGeek4Life, 2018-09-06:
Well, that's what I get for commenting without double-checking... I thought I had tested that and gotten errors or empty data, but I was apparently confusing it with what happens with centroided ITMS data.

chambm, 2018-09-12:
@alchemistmatt When you get a chance, see if the latest quameter build fixes the `nan` output.

alchemistmatt, 2018-09-13:
I tested four .raw files against [Build 1008](http://teamcity.labkey.org/viewLog.html?buildId=629360&buildTypeId=Bumbershoot_Windows_X86_64&tab=buildResultsDiv) and they all agreed perfectly with the January 2016 build of Quameter.  And, yes, the `nan` is gone, replaced with the original value.

  | Old | 9/4/2018 | 9/11/2018
-- | -- | -- | --
XIC-WideFrac | 0.115 | 0.148 | 0.115
XIC-FWHM-Q1 | 10.789 | 14.243 | 10.785
XIC-FWHM-Q2 | 12.794 | 17.449 | 12.790
XIC-FWHM-Q3 | 19.355 | 23.440 | 19.344
XIC-Height-Q2 | 0.993 | 1.068 | 0.993
XIC-Height-Q3 | 0.762 | 1.210 | 0.761
XIC-Height-Q4 | 3.373 | 5.356 | 3.374
RT-Duration | 5997.8 | 6000.7 | 6000.7
RT-TIC-Q1 | 0.299 | 0.128 | 0.299
RT-TIC-Q2 | 0.158 | 0.535 | 0.158
RT-TIC-Q3 | 0.092 | 0.109 | 0.092
RT-TIC-Q4 | 0.451 | 0.228 | 0.451
RT-MS-Q1 | 0.254 | 0.254 | 0.254
RT-MS-Q2 | 0.237 | 0.237 | 0.237
RT-MS-Q3 | 0.248 | 0.248 | 0.248
RT-MS-Q4 | 0.26 | 0.260 | 0.260
RT-MSMS-Q1 | 0.254 | 0.254 | 0.254
RT-MSMS-Q2 | 0.237 | 0.237 | 0.237
RT-MSMS-Q3 | 0.248 | 0.248 | 0.248
RT-MSMS-Q4 | 0.26 | 0.260 | 0.260
MS1-TIC-Change-Q2 | 0.957 | 0.981 | 0.957
MS1-TIC-Change-Q3 | 0.958 | 1.066 | 0.958
MS1-TIC-Change-Q4 | 3.69 | 2.217 | 3.690
MS1-TIC-Q2 | 0.613 | -nan(ind) | 0.613
MS1-TIC-Q3 | 0.673 | 0.555 | 0.673
MS1-TIC-Q4 | 1.099 | 0.590 | 1.099
MS1-Count | 1954 | 1954 | 1954
MS1-Freq-Max | 0.438 | 0.438 | 0.438
MS1-Density-Q1 | 2121 | 14590 | 2121
MS1-Density-Q2 | 2548 | 17536 | 2548
MS1-Density-Q3 | 2870 | 19864 | 2870
MS2-Count | 25402 | 25415 | 25415
MS2-Freq-Max | 5.729 | 5.729 | 5.729
MS2-Density-Q1 | 31 | 195 | 31
MS2-Density-Q2 | 58 | 377 | 58
MS2-Density-Q3 | 105 | 701 | 105

closed 2018-09-20 by alchemistmatt