ProteoWizard/pwiz #193: Metadata errors with new RawFileReader MSConvert

FarmGeek4Life, 2018-08-31:
MSConvert Version (x64):  3.0.18243

We just ran a test with the new MSConvert that uses RawFileReader, and noticed a few errors comparing older results vs. new results when converting to mzML. The peak data is the same, as is most of the metadata, but I'm going to add replies to this with the issues we have found.

To start off:
1. MS1 spectra have precursors displayed in the mzML, with default values for the items required under the precursorList.
2. HCD is being interpreted as ECD+CID. It appears to be caused by the pwiz Thermo [ActivationType Enum](https://github.com/ProteoWizard/pwiz/blob/master/pwiz_aux/msrc/utility/vendor_api/thermo/RawFileTypes.h#L644) using flag values and [directly casting](https://github.com/ProteoWizard/pwiz/blob/master/pwiz_aux/msrc/utility/vendor_api/thermo/RawFile.cpp#L1365) from the RawFileReader ActivationType enum, which uses sequential values (RawFileReader: HCD=5, RawFileTypes.h: CID=1, ECD=4)

alchemistmatt, 2018-08-31:
For reference, the enum values for ActivationType  in `ThermoFisher.CommonCore.Data.FilterEnums` are:
```
0: CollisionInducedDissociation
1: MultiPhotonDissociation
2: ElectronCaptureDissociation
3: PQD
4: ElectronTransferDissociation
5: HigherEnergyCollisionalDissociation (hcd)
6: Any
7: SAactivation
8: ProtonTransferReaction
9: NegativeElectronTransferDissociation
10: NegativeProtonTransferReaction
11: UltraVioletPhotoDissociation
```
This enum is returned by the call to `GetActivation` as demonstrated by this code snippet
```csharp
var scanFilter = mXRawFile.GetFilterForScanNumber(scan);
var reactions = scanFilter.MassCount;

var index = reactions - 1;
if (index > 0 && scanFilter.GetIsMultipleActivation(index))
{
    // The last activation is part of a ETciD/EThcD pair
    index--;
}

var activationTypeCode = scanFilter.GetActivation(index);
```
This snippet is from 
https://github.com/PNNL-Comp-Mass-Spec/Thermo-Raw-File-Reader/blob/ThermoNuGetDLL/XRawFileIO.cs#L1130

chambm, 2018-08-31:
Thanks for the report. I can definitely see your #2 happening because it happened with a few of the other enums when I started trying to cast directly (with MSFR all those are parsed from strings). It slipped by because apparently none of the pwiz or Skyline tests use HCD! Maybe I can figure out a way to adapt the ScanFilterTest tests to also test RawFileReader.

I cannot reproduce your #1. Can you provide a test file?

FarmGeek4Life, 2018-08-31:
You may be able to create a scan filter from a string with RawFileReader.
I think ThermoFisher.CommonCore.RawFileReader.FilterParserFactory.CreateFilterParser().GetFilterFromString(string) would do it.

FarmGeek4Life, 2018-08-31:
I also noticed that the "GetAverageMassList" method is now commented out, I'm assuming because it isn't used, but earlier this week I figured out how to get similar results from RawFileReader. I say "similar" because the method in MSFileReader is susceptible to the improper centroiding of FTMS data that would also occur with "GetMassList"

FarmGeek4Life, 2018-08-31:
Also, we just ran a test with a Lumos file with EThcD spectra; it crashed. The main error was "precursor count does not match ms level[SpectrumWorkerThreads::work] error in thread". I'm assuming that, if the creating a IScanFilter object using the GetFilterFromString works, this error will probably be fixed in the process of getting the ScanFilterTests to all pass.

FarmGeek4Life, 2018-08-31:
Also, if you call IScanFilter.ToString(), you get the scan filter string you would expect to see from MSFileReader.

chambm, 2018-08-31:
That was a pretty severe bug. No idea how widespread that was. Fixed in #194

chambm, 2018-09-05:
The activation type issue should be fixed in #204.

chambm, 2018-09-06:
(if someone finds other RFR issues it can be reopened)

closed 2018-09-06 by chambm