ProteoWizard/pwiz #329: Issues compiling on Windows with VS2013

rfellers, 2018-11-19:
When working on #165 in August, I was able to build the master branch without issues on VS2013. Now that @chambm has added 64 bit support for Thermo RAW files using RawFileReader, I'd like to pick my PR back up and finish. However, I can no longer build. I am getting a ton of **noexcept** errors like the following:
`.\pwiz/utility/misc/Filesystem.hpp(75) : error C3646: 'noexcept' : unknown override specifier`

A quick look seems to indicate that this keyword wasn't supported on Windows until VS2015 (https://msdn.microsoft.com/en-us/library/wfa0edys.aspx?f=255&MSPPError=-2147217396). Has the preferred Windows build setup changed? I am running the following command from PowerShell.

`.\quickbuild.bat -j4 --toolset=msvc-12.0 --i-agree-to-the-vendor-licenses address-model=64 msconvert msbenchmark`

Thanks,
Ryan

chambm, 2018-11-19:
You'll need VS2017 to build now. Community edition works fine.

rfellers, 2018-11-19:
Yup, VS2017 builds just fine for me, thanks. Sorry if I missed this announcement somewhere. I was a bit confused because the README.md stills mentions VS2013 in a couple places.

closed 2018-11-19 by rfellers

chambm, 2018-11-19:
Good point. Just fixed that.