ProteoWizard/pwiz #331: Wine linux version

rsalz, 2018-11-20:
Hello! On your readme it says that there is a wine linux version (in a docker container?). I just spent half of today trying to get msconvert to work in a wine env in ubuntu so I'm very interested! How can I obtain it?

chambm, 2018-11-20:
There's an older version here on DockerHub: https://hub.docker.com/r/chambm/pwiz-skyline-i-agree-to-the-vendor-licenses/

Eventually I'll be updating that one fairly frequently but for now you'll need to download the tarball and do `docker load` (maybe you can `docker load` from a URL, I'm not sure):
http://teamcity.labkey.org/guestAuth/repository/download/ProteoWizardAndSkylineDockerContainerWineX8664/lastSuccessful/pwiz-skyline-i-agree-to-the-vendor-licenses.tbz2

closed 2019-02-22 by chambm

soerendip, 2019-08-01:
Is there any description available on how to use the container?

chambm, 2019-08-01:
The latest version with tips on how to run it is at: https://hub.docker.com/r/proteowizard/pwiz-skyline-i-agree-to-the-vendor-licenses/