ProteoWizard/pwiz #360: Waters synapt2 mobility data: failure to read the scan number

lopippo, 2018-12-07:
Greetings, 

with the last version (downloaded and built today) but with previous ones also, I fail to get the scan number out of the mzML file generated starting from Water Synapt2 raw data. I succeed in getting that scan number when reading Thermo Orbitrap data.

This is what I code:

 pwiz::msdata::CVID native_id_format = pwiz::msdata::id::getDefaultNativeIDFormat(*mp_msDataFile);

(Note that the value that is returned is indeed for Waters files, so that id format is registered in pwiz)

Then, for each spectrum I iterate in, I do:

std::size_t scan_num = QString(pwiz::msdata::id::translateNativeIDToScanNumber(
                  native_id_format, spectrum->id).c_str()).toULong();

But this does not provide proper scan_num values, I systematically get  a value of 0.

I saw in the documentation that the getDefaultNativeIDFormat only works for selected file formats, and I could see that Waters Synapt2 files were not supported.

My question: since what I want is an absolutely unique identifier for the various mass spectra in the mzML file, I thought that I could standardize on the spectrum index (that seems to be more general than the scan number in the various file formats). 

Am I right if I say that there is *always* the same number of scans in a mzML file as there are spectra? If so, then I will be able to retrieve a mass spectrum from a mzML file using its index with the following call: spectrum (size_t index, bool getBinaryData=false). True or False?

Thank you for your kind attention,
Cheers,
Filippo

chambm, 2018-12-07:
The unique identifier for spectra (within a run, or mzML) is the nativeID, which is a string. Some nativeID formats can be converted to simple integers, but Waters and Sciex cannot. Index is a very bad identifier because it will change if any kind of filtering is done to the file (e.g. keep only MS1s, only MS2s, only ETD). NativeID will stay the same despite those filters. You can use an abbreviated form of the id string, like "1.0.1" for Waters instead of "function=1 process=0 scan=1". ProteoWizard has functions in the msdata::id namespace to serialize to/from these abbreviated ids.

lopippo, 2018-12-07:
chambm: thank you for your answer. Now I see how the index is a very bad idea!

Yes, I saw the abbreviate() function. But then how would one be sure that the values in the returned string would contain the scan number (and which one of the values, by the way)? 

So the first question to ask is: is there *systematically* a "scan=xx" string subpart to the ID string ? If so could one craft a simple regex to extract that value scan number? What does the mzML standard say about this ?

In other words, is there a sure way to get the scan number of each <spectrum> element in a mzML file, that does not depend on the peculiarities of the proprietary formats?

Thank you again for your attention,

Cheers,
Filippo

chambm, 2018-12-07:
The nativeID is not there just to make your life harder. It's a string, not a single number, because a single number cannot uniquely identify a spectrum in a WIFF file or a Waters RAW directory. For example, a WIFF can have multiple samples (runs) in it, and each sample can have multiple periods (theoretically). Each sample/period pair can have 1-N cycles, and each cycle can have 1-N experiments (like scan events in Thermo files). All 4 numbers are part of the unique id. There's no way to get a single number without some hacky encoding scheme where each component would be limited to a small range.

lopippo, 2018-12-10:
Greetings, Matthew, thanks for your patience and details. I think that I now have understood the situation. 
Cheers, Filippo

closed 2018-12-10 by lopippo

lopippo, 2019-02-18:
Greetings, Matt,

On Fri, Dec 07, 2018 at 12:00:54PM -0800, Matt Chambers wrote:
> The nativeID is not there just to make your life harder. It's a string, not a
> single number, because a single number cannot uniquely identify a spectrum in a
> WIFF file or a Waters RAW directory. For [...]

I would like to implement the mzML support in my program using libpwiz. When you
state that a WIFF can have multiple samples (runs), you presumably mean multiple
pwiz::msdata::Run? If so, I tried to find, in the documentation, the location of
the function that parses these runs. In MSData, I see a single Run data member.

Where can I find the code that allows one to know how many runs there are in a
given mzML file?

Thank you for your kind attention,
Sincerely,
Filippo

-- 

⢀⣴⠾⠻⢶⣦⠀  Filippo Rusconi, PhD
⣾⠁⢠⠒⠀⣿⡁   Scientist at CNRS
⢿⡄⠘⠷⠚⠋⠀   Debian Developer
⠈⠳⣄⠀⠀⠀⠀  http://msxpertsuite.org
          http://www.debian.org

chambm, 2019-02-18:
There is an overload in the Reader class which you feed a vector<MSDataPtr> objects. That will return all the samples in a WIFF file, each one as an MSData object with a single Run object. Alternately if you want a single run index there is an overload for specifying the run index as an integer. There is also a function to get the list of sample names rather than getting the full MSData objects.

lopippo, 2019-02-18:
On Mon, Feb 18, 2019 at 10:16:16AM -0800, Matt Chambers wrote:
> There is an overload in the Reader class which you feed a vector<MSDataPtr>
> objects. That will return all the samples in a WIFF file, each one as an MSData
> object with a single Run object. [...]

Awesome!

Thanks so much for the quick pointers!

Filippo
-- 

⢀⣴⠾⠻⢶⣦⠀  Filippo Rusconi, PhD
⣾⠁⢠⠒⠀⣿⡁   Scientist at CNRS
⢿⡄⠘⠷⠚⠋⠀   Debian Developer
⠈⠳⣄⠀⠀⠀⠀  http://msxpertsuite.org
          http://www.debian.org