ProteoWizard/pwiz #372: Build failure on mingw64

lopippo, 2018-12-12:
Greetings, Fellow Developers,

I build libpwiz on MS Windows (7 and 10) using the excellent MSYS2/MINGW64 development environment that mimick a GNU/Linux machine.

When I build the 3.0.18342 version in this environment with gcc 8.2.1 20181207, I get the following errors:

pwiz/utility/misc/Filesystem.cpp: In function 'void* {anonymous}::GetLibraryProcAddress(PSTR, PSTR)':
pwiz/utility/misc/Filesystem.cpp:151:30: error: invalid conversion from 'FARPROC' {aka 'long long int (*)()'} to 'PVOID' {aka 'void*'} [-fpermissive]
         return GetProcAddress(GetModuleHandleA(LibraryName), ProcName);
                ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

pwiz/utility/misc/Filesystem.cpp:273:27: error: reference to 'SYSTEM_HANDLE_INFORMATION' is ambiguous
     DWORD dwSize = sizeof(SYSTEM_HANDLE_INFORMATION);
                           ^~~~~~~~~~~~~~~~~~~~~~~~~
pwiz/utility/misc/Filesystem.cpp:102:12: note: candidates are: 'struct {anonymous}::SYSTEM_HANDLE_INFORMATION'
     struct SYSTEM_HANDLE_INFORMATION {
            ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from pwiz/utility/misc/Filesystem.cpp:35:
C:/msys64/mingw64/x86_64-w64-mingw32/include/winternl.h:837:5: note:                 'typedef struct _SYSTEM_HANDLE_INFORMATION SYSTEM_HANDLE_INFORMATION'
   } SYSTEM_HANDLE_INFORMATION, *PSYSTEM_HANDLE_INFORMATION;
     ^~~~~~~~~~~~~~~~~~~~~~~~~
pwiz/utility/misc/Filesystem.cpp:301:35: error: reference to 'SYSTEM_HANDLE_INFORMATION' is ambiguous
     auto pInfo = reinterpret_cast<SYSTEM_HANDLE_INFORMATION*>(pInfoBytes.data());
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
pwiz/utility/misc/Filesystem.cpp:102:12: note: candidates are: 'struct {anonymous}::SYSTEM_HANDLE_INFORMATION'
     struct SYSTEM_HANDLE_INFORMATION {
            ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from pwiz/utility/misc/Filesystem.cpp:35:
C:/msys64/mingw64/x86_64-w64-mingw32/include/winternl.h:837:5: note:                 'typedef struct _SYSTEM_HANDLE_INFORMATION SYSTEM_HANDLE_INFORMATION'
   } SYSTEM_HANDLE_INFORMATION, *PSYSTEM_HANDLE_INFORMATION;
     ^~~~~~~~~~~~~~~~~~~~~~~~~
pwiz/utility/misc/Filesystem.cpp:301:60: error: expected '>' before '*' token
     auto pInfo = reinterpret_cast<SYSTEM_HANDLE_INFORMATION*>(pInfoBytes.data());
                                                            ^
pwiz/utility/misc/Filesystem.cpp:301:60: error: expected '(' before '*' token
     auto pInfo = reinterpret_cast<SYSTEM_HANDLE_INFORMATION*>(pInfoBytes.data());
                                                            ^
                                                            (
pwiz/utility/misc/Filesystem.cpp:301:61: error: expected primary-expression before '>' token
     auto pInfo = reinterpret_cast<SYSTEM_HANDLE_INFORMATION*>(pInfoBytes.data());
                                                             ^
pwiz/utility/misc/Filesystem.cpp:301:81: error: expected ')' before ';' token
     auto pInfo = reinterpret_cast<SYSTEM_HANDLE_INFORMATION*>(pInfoBytes.data());
                                                                                 ^
                                                                                  )
make: *** [Makefile:5079: pwiz/utility/misc/Filesystem.lo] Error 1

Some errors seem to be specific of the MSWindows platform (FARPROC, I guess).

Could somebody be kind enough to help me find light here ?

Thank you for your attention,
Cheers,
Filippo

chambm, 2018-12-14:
Unfortunately we just don't officially support mingw64 since Visual Studio Community is available for free. But if you want to submit a PR that will disable the file handling code above (make it a no-op) when compiling with Mingw GCC, I'd accept that.

lopippo, 2018-12-15:
Ok, thanks,
I'll look into it more closely. Closing then.
Cheers,
Filippo

closed 2018-12-15 by lopippo