ProteoWizard/pwiz #409: Intensity scale changes of Sciex data after centroiding data

rickhelmus, 2019-01-07:
Hello,

When I use msconvert to convert TripleTOF 5600 Sciex data (GUI or command line, using a recent build) to mzML files the intensity scale seems to change to much lower values (several orders of magnitude):
[image]
Interestingly, the TIC/BPC values reported for each scan seem to be unaffected.

If I use `qtofpeakpicker` instead of vendor centroiding this doesn't occur. Does anyone else have experience with this?

Thanks,
Rick

qiangziMS, 2019-01-25:
I've also experienced this problem, and not only the scale changed, the ratio of peaks also changed. It seems like a version problem, when use old version of msconvert, every thing is OK ...

rickhelmus, 2019-01-25:
Hmm interesting! Any way you could share your old version? Unless I am missing something it seems that TeamCity doesn't s retain older versions...

chambm, 2019-02-28:
Vendor centroiding with Sciex uses area as the intensity rather than apex height. The Sciex people consider this more accurate and we changed it at their request. The centroid x value is certainly not the same as the apex x value. This was changed back in 2016.

CWT peak picker currently always gives the apex x value and intensity (and not even an interpolated apex of the wavelet, it's whatever the closest datapoint is).

closed 2019-02-28 by chambm