ProteoWizard/pwiz #461: empty sourceFileList violates schema

sneumann, 2019-03-05:
Hi, writing an MSData object without sourceFiles results in 
```
      <sourceFileList count="0">
      </sourceFileList>
```
which violates the schema (I wrongly reported that in https://github.com/HUPO-PSI/mzML/issues/2,
and it is biting us in https://github.com/sneumann/mzR/issues/192)

I think the `<` `sourceFileList` `>` should be skipped 
in https://github.com/ProteoWizard/pwiz/blob/master/pwiz/data/msdata/IO.cpp#L462
if `count==0`. I'd happily create a PR, but wouldn't know how to add a good test case to IOTest.cpp.

Yours, Steffen

chambm, 2019-03-05:
There are a lot of ways that you can create syntactically or semantically invalid files with MSData. It definitely relies on the client code knowing how to create a complete, valid document. In this case, I don't think we expected that a document could be created without sourceFiles, but I admit that ignores the possibility of creating MSData files _in silico_ or even straight from the instrument. I assume this is an _in silico_ case?

chambm, 2019-03-05:
https://github.com/ProteoWizard/pwiz/pull/462

closed 2019-03-11 by chambm