ProteoWizard/pwiz #483: Inaccurate MS2 precursor value = center of isolation window

mzdouc, 2019-03-19:
### Setup

I am reporting a problem with MsConvert from Proteowizard  version 3.0.19077-506c48e9c 64bit and would like to ask for help with this issue. 

I am using Windows 10 64bit as operating system.

The MS data was acquired on a Bruker qTOF, calibrated with an internal standard. 

### Problem description

When exporting Bruker .d-files with MsConvert, the resulting open format files (mzML, mzXML) display imprecise MS2 precursor values that do not match the values in the corresponding MS1 scans:

[image]

Interestingly, the displayed precursor value of scan #2160 is exactly the centre of the isolation window.

However, in the original data, the values are calibrated and correct: 

[image]

This behaviour can be be reproduced using the command-line version of MsConvert. 

The paramerters used in both GUI and command line were: 
PeakPicking: vendor, MS-level=1-2
Binary encoding precision:  32bit
Output format: mzML

Applying the **precursorRecalculation** filter in the command line version of msconvert was not successful:

```
C:\Program Files\ProteoWizard\ProteoWizard 3.0.19077.506c48e9c>msconvert.exe d:6132_6136_BD3_01_31134.d --32 --filter "peakPicking [vendor[msLevel=1-2]]" --filter "precursorRecalculation" -o d:
format: mzML
    m/z: Compression-None, 32-bit
    intensity: Compression-None, 32-bit
    rt: Compression-None, 32-bit
ByteOrder_LittleEndian
 indexed="true"
outputPath: d:
extension: .mzML
contactFilename:
runIndexSet:

spectrum list filters:
  peakPicking [vendor[msLevel=1-2]]
  precursorRecalculation

chromatogram list filters:

filenames:
  d:6132_6136_BD3_01_31134.d

processing file: d:6132_6136_BD3_01_31134.d
calculating source file checksums
Error writing run 1 in "6132_6136_BD3_01_31134.d":
[SpectrumList_PrecursorRecalculator] Mass analyzer not supported: time-of-flight
```

Interestingly, also Bruker's Compass Export tool produced the same problem (inaccurate MS2 precursor values).

I would appreciate any suggestions on how to solve this problem. I have no clue how to proceed from this point except for ignoring the inaccuracy.

The original file can be obtained upon request. 

Thank you for your help!

chambm, 2019-03-19:
Is this .baf format data? If so, there should be an analysis.sqlite file in the .d directory (after you've accessed it once with ProteoWizard). Attach that file and I'll see if there's any way to get the "correct" isolation m/z from the metadata in that file. I suspect not, but it won't hurt to take a second look.

mzdouc, 2019-03-19:
Hey @chambm 

thanks for the quick reply!

The files are in fact in .baf-format - for the analysis.sqlite file, see following link:

https://www.dropbox.com/s/f28595s56o4mcei/analysis.sqlite?dl=0

You can drop me a line at mzdouc (a ) naicons.com if you want to take a look at the whole directory.

chambm, 2019-03-19:
Sorry, the "correct" isolation m/z isn't in the sqlite file. Thus it's not a property provided by the vendor API. It's unfortunate precursorRecalculation doesn't work, but it was designed for Thermo Orbi/FT. Try precursorRefinement. Looks like it supports TOF.

mzdouc, 2019-03-20:
@chambm 
I tried to apply the **precursorRefine** filter on my data but it did not change anything:

```
C:\Program Files\ProteoWizard\ProteoWizard 3.0.19077.506c48e9c>msconvert.exe c:\Users\mzdou\Desktop\6132_6136_BD3_01_31134.d -o c:\Users\mzdou\Desktop\  --32 -filter "peakPicking[cwt]" "precursorRefine" -v
[msconvert] no files found matching "peakPicking[cwt]"
[msconvert] no files found matching "precursorRefine"
format: mzML
    m/z: Compression-None, 32-bit
    intensity: Compression-None, 32-bit
    rt: Compression-None, 32-bit
ByteOrder_LittleEndian
 indexed="true"
outputPath: c:\Users\mzdou\Desktop\
extension: .mzML
contactFilename:
runIndexSet:

spectrum list filters:

chromatogram list filters:

filenames:
  c:\Users\mzdou\Desktop\6132_6136_BD3_01_31134.d

processing file: c:\Users\mzdou\Desktop\6132_6136_BD3_01_31134.d
calculating source file checksums
writing output file: c:\Users\mzdou\Desktop\6132_6136_BD3_01_31134.mzML
converting spectra: 5274/5274
```

This output is strange, because it seems like the program is ignoring the precursorRefine parameter.

Anyway, I have also contacted Bruker and asked for advice. If we manage to find a solution, I will post it here as well.

Thanks for you help and effort!

chambm, 2019-03-21:
You put:
`-filter "peakPicking[cwt]" "precursorRefine"`
You need:
`--filter "peakPicking cwt" --filter "precursorRefine"`

closed 2019-03-21 by chambm