ProteoWizard/pwiz #509: correct cmg parameters

xuesu, 2019-04-05:
Excuse me, I wonder how to start this process in cmd in order to run it on linux. 
I have tried some parameters, like "C:\Program Files\ProteoWizard\ProteoWizard 3.0.11856\msconvert.exe" --filter "peakPicking true 1-" -z *.wiff
however, the outputs are not exactly the same as those generated by GUI,
Thank you for your time!!
[image]
[image]

chambm, 2019-04-05:
Huh, that second picture is running on Linux? The version doesn't look the same though.

xuesu, 2019-04-05:
Thanks for your time! I have different versions of pwiz on my computers, so I capture the snap of both of them. Only settings with red line matter.

chambm, 2019-04-05:
And what is the command-line you're using that you're saying does not give equivalent results? And what exactly do you mean by "not exactly the same" - what's different about the results?

xuesu, 2019-04-06:
Hmm...Using GUI, I got a mzml of 1.2GB, with informations of ms level, using only cmd, I got a mzml of 600mb without ms level.
What I wanna do is to understand what the correct parameters are and then use them on linux. Now I just tried them on win.

chambm, 2019-04-12:
What's the exact command-line you used? I could imagine TPP compatibility could be a huge bloat factor if you're working with IMS data (millions of sparse spectra).