ProteoWizard/pwiz #568: header/metadata from Waters raw files

stanstrup, 2019-06-11:
Hello,

I am working on a project to automatically do some diagnostics on files as soon as a run finishes.
The problem I am running into is getting relevant metadata from the converted file.
For example I need to know the file description (in the Masslynx samplelist) and the project name (in our lab I can only get that from the samplelist name).

The info I need is available in the `_HEADER.TXT` file but is not added to the converted files meta data.
The lines I am interested in is:
```
$$ Job Code: Some project
$$ Sample Description: Important sample
```

I was looking through the mzML specs and examples and from what I could gather a possible solution would be to add something like:

```
    <sampleList count="1">
      <sample id="org_filename.raw" name="Important sample">
     <userParam name="Job Code" value="Some project"/>
      </sample>
    </sampleList>
```
Or alternatively:
```
    <sampleList count="1">
      <sample id="org_filename.raw" name="org_filename.raw">
     <userParam name="Job Code" value="Some project"/>
     <userParam name="Sample Description" value="Important sample"/>
      </sample>
    </sampleList>
```

Would this be something you could consider?
If not, would there be anything wrong about injecting the above into the file?

chambm, 2019-06-12:
I think this would be fine: PR welcome. :) But I would go with:

> <sampleList count="1">
      <sample id="<sample description>">
     <userParam name="Job Code" value="<job code>"/>
        <cvParam cvRef="MS" accession="MS:1000002" name="sample name" value="<sample description>"/>
      </sample>
    </sampleList>

Basically copy the code from Reader_Thermo.cpp:145 but change where you get the sample id.

stanstrup, 2019-06-12:
A PR to non-R code is unfortunately outside my capability.
Your suggestion does indeed look better. From reading the specs though I think you'd need to ensure that the sample id is unique in case of multiple files.