ProteoWizard/pwiz #590: MSConvert batch processing issue

saersamani, 2019-07-24:
Hi, 

When I run the latest version of MSConvert GUI on a Windows 7 64 bit on SWATH data, some of the scans remain empty with no recorded data. I do not have this issue when convert the files one at the time. Do you know what could cause such a problem?

Cheers,
Saer

chambm, 2019-07-24:
Can you attach some screenshots demonstrating the problem? By "scan remain empty" do you mean an entire file, or just a few scans within the file?

saersamani, 2019-07-24:
Please see two screenshots below. The first one shows the used 
parameters. and the second one shows the same files converted 
individually (folder called "mzXML_Files") and as a batch (folder called 
"Raw_Data_Pos"). As you can see the file "EA_001A" is completely empty 
in the batch mode and also the other files do not record all the scans.

chambm, 2019-08-23:
I think you'll need to visit the GitHub issue directly to post attachments. The screenshots didn't come through.

saersamani, 2019-08-24:
[image]
[image]

Here are the screenshots.

saersamani, 2019-08-26:
> The "note: the input only contains chromatogram data" message means you should use the "SIM as Spectra" and/or "SRM as Spectra" options. How were the files in the "mzXML" directory created?

The first screenshot shows the parameter settings for the conversion. As you can see in that picture, none of those were checked. I will try again with “SIM as Spectra” and will let you know.

saersamani, 2019-08-27:
Checking the SIM option solved the problem. Thanks for the help.

closed 2019-08-27 by saersamani

sorenwacker, 2022-06-11:
How can I load multiple file in there?