ProteoWizard/pwiz #606: msconvert command line tool missing on Windows 10 Pro

fkromer, 2019-08-09:
I downloaded [Proteowizard, Windows 64-bit Installer (able to convert vendor files except T2D)](http://www.proteowizard.org/download.html) and installed it on Windows 10 Pro. After installation the MSConvertGUI works just fine. But I miss the [`msconvert` command line tool](http://www.proteowizard.org/tools/msconvert.html). Where is the command line tool located and how can I run it via e.g. `command prompt`?

chambm, 2019-08-09:
The command-line msconvert.exe is always installed in the same directory as the GUI, but there is no shortcut created to it like there is for the GUI, and it's not added to your system or user PATH (which is where it would have to be in order for you to just be able to call `msconvert blah blah` on the command prompt and have it find it).

Where it's located depends on whether you installed per-user or per-machine. For per-machine it's in the `Program Files/ProteoWizard/ProteoWizard <version>` directory. For per-user it's under `%LOCALAPPDATA%\Apps\ProteoWizard <version>`

closed 2019-08-09 by chambm

fkromer, 2019-08-09:
I totally missed that cause most Windows installers add the required paths to PATH during installation implicitly. Thanks.