ProteoWizard/pwiz #607: Clarification of msconvert command line interface

fkromer, 2019-08-09:
According to [msconvert outputs help into stderr](https://github.com/ProteoWizard/pwiz/issues/543) `msconvert` pipes the help via stderr (instead of via stdout). Is the command line interface despite of this design decision implemented like one would usually expect? Is stderr used for all output? Are there return codes? What do the return codes mean?

chambm, 2019-08-09:
All status/note/warning/error output except one warning message (which is just an oversight) will go to stderr, yes. If the return code is greater than 0, there was at least 1 error. If it's greater than 1, then it means the number of files that failed to convert. If it's 1 it could just be one file failed to convert or something else was wrong with the command.

fkromer, 2019-08-09:
Thanks for clarification. I stumbled over this topic cause I am trying to call `msconvert` via Python `subprocess` or a `subprocess` wrapper and wondered why command line app wrapping is not working as usual 😄 Have you some recommendations w.r.t. this topic?

fkromer, 2019-08-14:
> Have you some recommendations w.r.t. this topic?

I implemented an ugly workaround solution but it's working for me right now.

fkromer, 2019-08-14:
Another note: In case a `.mzML` file is already existing the file is overridden.

closed 2019-08-14 by fkromer