ProteoWizard/pwiz #617: Version releases

vanzod, 2019-08-23:
We received a request to install `pwiz` in the software stack of our HPC cluster. For reproducibility reasons we require all the installed software releases to be identifiable through a version number.
Do you have any plan to start releasing versions? I am sure that this would definitely benefit not just us but the user community at large.

chambm, 2019-08-23:
Our software is released from our TeamCity CI system upon successful build but is always released with a unique version. The version is available from the command-line usage statement at the bottom:
`ProteoWizard release: 3.0.19227 (d3ef1d04b)`

The revision number is 2 digit year and day of year (1-365) and the hash is the last git commit included in the release.

Note that if you're receiving a request to install on HPC (presumably Linux), the user probably wants vendor reader support. In that case you need to use the [Wine image with Docker or Singularity](https://hub.docker.com/r/proteowizard/pwiz-skyline-i-agree-to-the-vendor-licenses) so you can run the Windows release on Linux.

vanzod, 2019-08-23:
@chambm Thank you for the clarification. What is the procedure to download an archive containing the sources for an older version?

chambm, 2019-09-10:
The source code is on GitHub. It should be straightforward to find a specific commit with the version hash (e.g. d3ef1d04b above) and download it from GitHub either through git or as a zip.

closed 2019-09-10 by chambm