ProteoWizard/pwiz #647: running in wine in singularity - 'Caught unknown exception'

joannawolthuis, 2019-09-23:
Hi team! I am struggling with a bit of a meta-meta problem - I want to run msconvert on a linux based HPC, that cannot run docker (fun to begin with.. its a permissions issue) and together with a colleague I have managed to install the docker container through singularity.
Now, I managed to get this command to work (once), but it does not anymore: when I try to run through singularity with a command akin to:

`singularity exec -B /my/home /my/home/pwiz_singularity_container wine msconvert /my/home/data/file.raw -o /my/home/data/converted_files –-mzXML`

I see the following error:
`[msconvert] no files found matching "–-mzXML"
format: mzML
    m/z: Compression-None, 64-bit
    intensity: Compression-None, 32-bit
    rt: Compression-None, 64-bit
ByteOrder_LittleEndian
 indexed="true"
outputPath: /my/home/data/converted/file.mzXML
extension: .mzML
contactFilename:
runIndexSet:

spectrum list filters:

chromatogram list filters:

filenames:
/my/home/data/raw_file.raw

processing file: 
/my/home/data/raw_file.raw
[C:\pwiz\msconvert.exe] Caught unknown exception.
Please report this error to support@proteowizard.org.
Attach the command output and this version information in your report:

ProteoWizard release: 3.0.19252 (aa45583de)
Build date: Sep  9 2019 07:06:22`

Please excuse me for not showing the actual file paths, its a bit of a privacy issue. 
Two problems - one: not recognising the --mzXML flag at all, it seems?
And two: the unknown exception. 

I'd love to include logs but I am not sure where to find them. Should I be having certain environment settings up? Since I'm in a wrapper in a wrapper, I hope there's still a fix.. and I hope I just missed something simple and it's fixable!

Thanks so much :)
Kind regards,
Joanna

chambm, 2019-09-23:
The --mzXML issue looks like a copy/paste error. One of the hyphens isn't a basic hypen.
`–-mzXML`
vs
`--mzXML`

The unknown exception is tougher. Does that file work with msconvert when running it in Windows? Is your singularity run able to convert other vendor files and/or other RAW files?

joannawolthuis, 2019-09-23:
Wew, thanks for your super fast reply! That solves the --mzXML issue, at least something is simple =) 
I'm going to try on other files, but I tried a few and it doesn't seem to work yet. Do you know of any 'reference' raw/etc files that are verified to be OK that I could download, by any chance? (In the meantime I'll try some more, but just in case)

joannawolthuis, 2019-09-23:
Ah, to add, I also get:

```
wineserver: could not save registry branch to system.reg : Read-only file system
wineserver: could not save registry branch to user.reg : Read-only file system
```
a few seconds after the initial error message.

chambm, 2019-09-23:
Ah that looks bad. I think the image needs to be read-write.

joannawolthuis, 2019-09-23:
I'm going to see if I can open it in read-write, will update ^^

chambm, 2019-09-23:
There are test files for each vendor:
https://github.com/ProteoWizard/pwiz/blob/master/pwiz/data/vendor_readers/Thermo/Reader_Thermo_Test.data.tar.bz2
https://github.com/ProteoWizard/pwiz/blob/master/pwiz/data/vendor_readers/Agilent/Reader_Agilent_Test.data.tar.bz2
etc.

joannawolthuis, 2019-09-23:
Thanks :) I tested on thermo file and i get the same unknown exception, guessing it's not the RAW file then. Still trying to figure out permissions, it seems to not be easy...

joannawolthuis, 2019-09-23:
Executing as 'singularity --writable' gives me a slightly different message:
```
[msconvert] no files found matching "/hpc/cog_bioinf/ridder/users/jwolthuis/Data/Metabolomics/test/Reader_Thermo_Test.data/090701-LTQVelos-unittest-01.raw"
[msconvert] No files specified.
```
although that file does exist - any ideas? I am not very familiar with the inner workings of singularity, so if it's more something I should tag their team about please let me know ; w;

joannawolthuis, 2019-09-23:
Hi there, to save you some effort :-) I think I found what's wrong, something with the bind folder for singularity isn't working and all I can access from within singularity is my home folder. I moved the raw file there and it converts without errors, so I'm going to sort out that problem and hopefully move on to the rest of the pipeline :D

closed 2019-12-03 by chambm

HeatherKates, 2022-02-03:
I know this issue is closed, but I ran into the same problem and fixed it by adding the preferred folder as a drive in the 'winecfg' > 'Drives' tab. This seems like a better long-term solution so I wanted to share it in case anyone else came to this issue via google!

chambm, 2022-02-03:
Can you elaborate on how you ran winecfg in a way that persisted the changes?

HeatherKates, 2022-02-03:
I'm not sure if I understand your question but realize I left out an important detail that likely addresses it - I did this on an HPC desktop session. I don't know how often this is an option on HPC clusters. This enables us to configure and run graphical user interface applications on our cluster. So when I use the command winecfg in a desktop session termainl, the "Wine configuration" window appears and I literally click the "Drives" tab and "Add..." and type the path to my drive. Without the HPC desktop session option I guess it would be done like https://forum.winehq.org/viewtopic.php?t=34039 but I haven't tested that.

chambm, 2022-02-03:
So you aren't actually using our Docker container or a Singularity derivative of it?

HeatherKates, 2022-02-03:
Oh, I'm sorry, you're correct. We had been in parallel trying to install and use proteowizard via the windows installer + wine and chambm/pwiz-skyline-i-agree-to-the-vendor-licenses using docker (singularity) and I got mixed up here. Sorry if this wasn't relevant.

chambm, 2022-02-03:
Which version of wine did you use? I can't upgrade the container to Wine 7 because it causes an infinite loop in Skyline's unit tests, and I can't upgrade Bruker's Baf2Sql API without upgrading Wine because it crashes with the current Wine version (3.12). Lots of other versions I tested had one or the other problem as well.

HeatherKates, 2022-02-03:
wine 7

chambm, 2022-02-03:
OK, good luck then. :) Wine 7 works fine with all the msconvert unit tests, at least.

chambm, 2026-01-21:
Are you using the current container that's Wine 10 IIRC?