ProteoWizard/pwiz #690: Error building on Windows

ghassett, 2019-10-23:
Hello, I am trying to build pwiz on Windows with Visual Studio 2019 Community Edition installed.  When I run `quickbuild.bat`, it terminates early with the following error.  Am I missing something obvious?  I usually work on Linux/Unix so please forgive me if there is something really obvious that I need to do prior to building.

Thanks very much for your help!

// greg //

```
tar.list expat-2.0.1.tar.bz2
tar.extract expat-2.0.1.tar.bz2 (already extracted)
[pwiz_tools/BiblioSpec] Changing code page to UTF-8 (65001)
tar.list inputs.tar.bz2
tar.extract inputs.tar.bz2 (already extracted)
error: Name clash for '<pC:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\BiblioSpec\src\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi>BlibBuild.exe'
error:
error: Tried to build the target twice, with property sets having
error: these incompatible properties:
error:
error:     -  none
error:     -  <assembly>object(file-target)@1290 <define>PWIZ_READER_ABI_T2D
error:
error: Please make sure to have consistent requirements for these
error: properties everywhere in your project, especially for install
error: targets.

Build finished at 17:50:40.77
Elapsed time: 0:0:46
```

chambm, 2019-10-23:
Try just building msconvert: add "msconvert" to the command-line. Also to get vendor support you must add `--i-agree-to-the-vendor-licenses`

Try
`quickbuild --help` as well

ghassett, 2019-10-23:
Thanks for the rapid response, Matt -- very much appreciated.
Getting a lot further now!  Looks like the build is going to take a while.

Once I build msconvert via `quickbuild msconvert --i-agree-to-the-vendor-licenses`, is it possible to build from within the Visual Studio IDE, to step through the code and learn my way around?

chambm, 2019-10-23:
Oh, add -j<num cores to build in parallel>

ghassett, 2019-10-23:
Hi Matt,

Here's the tail of console output when I run `quickbuild.bat msconvert --i-agree-to-the-vendor-licenses debug-symbols=on` -- `msconvert.exe` is not created (fails at link time, it appears):

```
...on 1000th target...
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\asynch-exceptions-on\debug-symbols-on\link-static\threading-multi\ThermoRawMetaDump.obj
msvc.link C:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\asynch-exceptions-on\debug-symbols-on\link-static\threading-multi\ThermoRawMetaDump.exe
LINK : fatal error LNK1181: cannot open input file 'pwiz_vendor_api_thermo.lib'

        call "C:\Users\ghassett\pwiz\build-nt-x86\standalone\msvc\msvc-14.2\msvc-setup.bat"  >nul
 link /NOLOGO /INCREMENTAL:NO /DEBUG /MACHINE:X86 /MANIFEST /OPT:REF,ICF /subsystem:console /out:"C:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\asynch-exceptions-on\debug-symbols-on\link-static\threading-multi\ThermoRawMetaDump.exe"   @"C:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\asynch-exceptions-on\debug-symbols-on\link-static\threading-multi\ThermoRawMetaDump.exe.rsp"
        if %ERRORLEVEL% NEQ 0 EXIT %ERRORLEVEL%

...failed msvc.link C:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\asynch-exceptions-on\debug-symbols-on\link-static\threading-multi\ThermoRawMetaDump.exe C:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\asynch-exceptions-on\debug-symbols-on\link-static\threading-multi\ThermoRawMetaDump.pdb...
...skipped <pC:\Users\ghassett\pwiz\build-nt-x86\msvc-release>ThermoRawMetaDump.exe for lack of <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\asynch-exceptions-on\debug-symbols-on\link-static\threading-multi>ThermoRawMetaDump.exe...
...skipped <pC:\Users\ghassett\pwiz\build-nt-x86\msvc-release>ThermoRawMetaDump.pdb for lack of <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\asynch-exceptions-on\debug-symbols-on\link-static\threading-multi>ThermoRawMetaDump.pdb...
...skipped <pC:\Users\ghassett\pwiz\build-nt-x86\msvc-release>msconvert.exe for lack of <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\commandline\msvc-14.2\release\asynch-exceptions-on\debug-symbols-on\link-static\threading-multi>msconvert.exe...
...failed updating 3 targets...
...skipped 5 targets...
...updated 1058 targets...
Build finished at 18:41:38.88
Elapsed time: 0:12:33
```

ghassett, 2019-10-24:
Tried a slightly different command:
```
quickbuild.bat --i-agree-to-the-vendor-licenses -j 2 msconvert
```
but still unable to build on Windows.  Tail of the build log:
```
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\process.h(44): error C2883: 'system': function declaration conflicts with 'boost::nowide::system' introduced by using-declaration
C:\Users\ghassett\pwiz\libraries\boost_aux\boost/nowide/system.hpp(28): note: see declaration of 'boost::nowide::system'
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\process.h(44): error C2375: 'boost::nowide::system': redefinition; different linkage
C:\Users\ghassett\pwiz\libraries\boost_aux\boost/nowide/system.hpp(28): note: see declaration of 'boost::nowide::system'

    call "C:\Users\ghassett\pwiz\build-nt-x86\standalone\msvc\msvc-14.2\msvc-setup.bat"  >nul
 cl /Zm800 -nologo @"C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj.rsp"  > "C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj.out"
    set status=%ERRORLEVEL%
    findstr /V /X "SpectrumList_Thermo.cpp" "C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj.out"
    exit %status%

...failed compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj...
...skipped <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi>libpwiz_reader_thermo.lib for lack of <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi>SpectrumList_Thermo.obj...
...skipped <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\commandline\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi>msconvert.exe for lack of <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi>libpwiz_reader_thermo.lib...
...skipped <pC:\Users\ghassett\pwiz\build-nt-x86\msvc-release>msconvert.exe for lack of <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\commandline\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi>msconvert.exe...
...failed updating 2 targets...
...skipped 4 targets...
...updated 10 targets...
Build finished at  8:51:27.18
Elapsed time: 0:0:23
```

chambm, 2019-10-24:
Hmm, add `address-model=64` to the command. I thought I'd made it the default but apparently not. But good to know that the default 32-bit isn't working on VS2019, will have to look into that.

chambm, 2019-10-24:
You can put it anywhere in the command except between -j and 2. It' a Boost.Build feature specification rather than an ARGV flag (starts with --) or Boost.Build option (starts with -). But any type of option can go in any position.

ghassett, 2019-10-24:
Thank you Matt -- here's my latest command line:
```
quickbuild.bat --i-agree-to-the-vendor-licenses -j 2 address-model=64 msconvert
```
and the tail of the build log (note that `msconvert.exe` is not created):
```
...on 900th target...
...on 1000th target...
...skipped <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\commandline\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi>msconvert.exe for lack of <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi>libpwiz_reader_thermo.lib...
...on 1100th target...
...on 1200th target...
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\ThermoRawMetaDump.obj
msvc.link C:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\ThermoRawMetaDump.exe
msvc.manifest C:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\ThermoRawMetaDump.exe
...skipped <pC:\Users\ghassett\pwiz\build-nt-x86\msvc-release-x86_64>msconvert.exe for lack of <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\commandline\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi>msconvert.exe...
...failed updating 1 target...
...skipped 3 targets...
...updated 1286 targets...
Build finished at 10:40:59.52
Elapsed time: 0:13:9

C:\Users\ghassett\pwiz>dir /s msconvert.exe
 Volume in drive C has no label.
 Volume Serial Number is 44B6-A28E
File Not Found

C:\Users\ghassett\pwiz>
```

chambm, 2019-10-24:
That doesn't include an error. Run the command again so it only tries to rebuild what didn't build last time and it should be a briefer log. But there should be a line starting with "...failed" in there somewhere.

ghassett, 2019-10-24:
Thanks Matt -- here is the error:
```
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\analysis\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\Version.obj
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\Version.obj
msvc.archive C:\Users\ghassett\pwiz\build-nt-x86\pwiz\analysis\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\libpwiz_analysis_version.lib
msvc.archive C:\Users\ghassett\pwiz\build-nt-x86\pwiz\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\libpwiz_version.lib
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\msdata\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\Version.obj
msvc.archive C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\msdata\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\libpwiz_data_msdata_version.lib
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\identdata\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\Version.obj
msvc.archive C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\identdata\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\libpwiz_data_identdata_version.lib
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\proteome\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\Version.obj
msvc.archive C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\proteome\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\libpwiz_data_proteome_version.lib
msvc.link C:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\ThermoRawMetaDump.exe
LINK : fatal error LNK1181: cannot open input file 'pwiz_vendor_api_thermo.lib'

        call "C:\Users\ghassett\pwiz\build-nt-x86\standalone\msvc\msvc-14.2\msvc-setup.bat"  >nul
 link /NOLOGO /INCREMENTAL:NO /MACHINE:X86 /MANIFEST /subsystem:console /out:"C:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\ThermoRawMetaDump.exe"  /OPT:REF,ICF @"C:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\ThermoRawMetaDump.exe.rsp"
        if %ERRORLEVEL% NEQ 0 EXIT %ERRORLEVEL%

...failed msvc.link C:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\ThermoRawMetaDump.exe...
...skipped <pC:\Users\ghassett\pwiz\build-nt-x86\msvc-release>ThermoRawMetaDump.exe for lack of <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\prototype\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi>ThermoRawMetaDump.exe...
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\process.h(44): error C2883: 'system': function declaration conflicts with 'boost::nowide::system' introduced by using-declaration
C:\Users\ghassett\pwiz\libraries\boost_aux\boost/nowide/system.hpp(28): note: see declaration of 'boost::nowide::system'
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\process.h(44): error C2375: 'boost::nowide::system': redefinition; different linkage
C:\Users\ghassett\pwiz\libraries\boost_aux\boost/nowide/system.hpp(28): note: see declaration of 'boost::nowide::system'

    call "C:\Users\ghassett\pwiz\build-nt-x86\standalone\msvc\msvc-14.2\msvc-setup.bat"  >nul
 cl /Zm800 -nologo @"C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj.rsp"  > "C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj.out"
    set status=%ERRORLEVEL%
    findstr /V /X "SpectrumList_Thermo.cpp" "C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj.out"
    exit %status%

...failed compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj...
...skipped <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi>libpwiz_reader_thermo.lib for lack of <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi>SpectrumList_Thermo.obj...
...skipped <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\commandline\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi>msconvert.exe for lack of <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi>libpwiz_reader_thermo.lib...
...skipped <pC:\Users\ghassett\pwiz\build-nt-x86\msvc-release>msconvert.exe for lack of <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\commandline\msvc-14.2\release\asynch-exceptions-on\link-static\threading-multi>msconvert.exe...
...failed updating 2 targets...
...skipped 4 targets...
...updated 10 targets...
Build finished at 11:02:55.46
Elapsed time: 0:0:25
```

chambm, 2019-10-24:
Hmm, you get that error with the command including `address-model=64`? Because that's trying to build the 32-bit version (the output paths don't have `adrs-mdl-64` in them).

ghassett, 2019-10-24:
Yes, the exact command was (copying from the console window):
```
C:\Users\ghassett\pwiz>quickbuild.bat --i-agree-to-the-vendor-licenses -j 2 --address-model=64 msconvert
```

ghassett, 2019-10-24:
I may have had the `--` before `address-model=64`.  Am retrying now...

chambm, 2019-10-24:
Yes I see I didn't answer your question above. It has to be without the dashes.

ghassett, 2019-10-24:
Here is the unedited output of the console window, including the command I used and the results:
```
C:\Users\ghassett\pwiz>quickbuild.bat --i-agree-to-the-vendor-licenses -j 2 address-model=64 msconvert
Build started at 11:12:07.98
Building pwiz (64-bit)...
ProteoWizard 3.0.19297.c8b8b470d master x64 AMD64
Generated C:\Users\ghassett\pwiz\pwiz\data\msdata\Version.cpp
Generated C:\Users\ghassett\pwiz\pwiz\analysis\Version.cpp
Generated C:\Users\ghassett\pwiz\pwiz\data\identdata\Version.cpp
Generated C:\Users\ghassett\pwiz\pwiz\data\tradata\Version.cpp
Generated C:\Users\ghassett\pwiz\pwiz\data\proteome\Version.cpp
Generated C:\Users\ghassett\pwiz\pwiz\Version.cpp
tar.list hdf5-1.8.7.tar.bz2
tar.extract hdf5-1.8.7.tar.bz2 (already extracted)
tar.list boost_1_67_0.tar.bz2
tar.extract boost_1_67_0.tar.bz2 (already extracted)
tar.list zlib-1.2.3.tar.bz2
tar.extract zlib-1.2.3.tar.bz2 (already extracted)
Extracting vendor APIs...
warning: No toolsets are configured.
warning: Configuring default toolset "msvc".
warning: If the default is wrong, your build may not work correctly.
warning: Use the "toolset=xxxxx" option to override our guess.
warning: For more configuration options, please consult
warning: http://boost.org/boost-build2/doc/html/bbv2/advanced/configuration.html
Warning: Not using Mascot Parser due to unsupported version of MSVC.
tar.list libpng-1.5.6.tar.bz2
tar.extract libpng-1.5.6.tar.bz2 (already extracted)
tar.list freetype-2.4.7.tar.bz2
tar.extract freetype-2.4.7.tar.bz2 (already extracted)
tar.list libgd-2.1.0alpha.tar.bz2
tar.extract libgd-2.1.0alpha.tar.bz2 (already extracted)
tar.list hdf5-1.8.7.tar.bz2
tar.extract hdf5-1.8.7.tar.bz2 (already extracted)
[pwiz/data/vendor_readers/ABI/T2D] 64-bit MSVC build: Reader_ABI_T2D can only identify files, not read them.
...patience...
...patience...
...patience...
...patience...
...patience...
...patience...
...patience...
...found 42146 targets...
...updating 14 targets...
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\Version.obj
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\analysis\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\Version.obj
msvc.archive C:\Users\ghassett\pwiz\build-nt-x86\pwiz\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\libpwiz_version.lib
msvc.archive C:\Users\ghassett\pwiz\build-nt-x86\pwiz\analysis\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\libpwiz_analysis_version.lib
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\msdata\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\Version.obj
msvc.archive C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\msdata\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\libpwiz_data_msdata_version.lib
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\identdata\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\Version.obj
msvc.archive C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\identdata\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\libpwiz_data_identdata_version.lib
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\proteome\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\Version.obj
msvc.archive C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\proteome\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\libpwiz_data_proteome_version.lib
compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\process.h(44): error C2883: 'system': function declaration conflicts with 'boost::nowide::system' introduced by using-declaration
C:\Users\ghassett\pwiz\libraries\boost_aux\boost/nowide/system.hpp(28): note: see declaration of 'boost::nowide::system'
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\process.h(44): error C2375: 'boost::nowide::system': redefinition; different linkage
C:\Users\ghassett\pwiz\libraries\boost_aux\boost/nowide/system.hpp(28): note: see declaration of 'boost::nowide::system'

    call "C:\Users\ghassett\pwiz\build-nt-x86\standalone\msvc\msvc-14.2\address-model-64\msvc-setup.bat"  >nul
 cl /Zm800 -nologo @"C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj.rsp"  > "C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj.out"
    set status=%ERRORLEVEL%
    findstr /V /X "SpectrumList_Thermo.cpp" "C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj.out"
    exit %status%

...failed compile-c-c++ C:\Users\ghassett\pwiz\build-nt-x86\pwiz\data\vendor_readers\Thermo\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\SpectrumList_Thermo.obj...
[... cut ...] el-64\asynch-exceptions-on\link-static\threading-multi>libpwiz_reader_thermo.lib...
...skipped <pC:\Users\ghassett\pwiz\build-nt-x86\msvc-release-x86_64>msconvert.exe for lack of <pC:\Users\ghassett\pwiz\build-nt-x86\pwiz_tools\commandline\msvc-14.2\release\address-model-64\asynch-exceptions-on\link-static\threading-multi>msconvert.exe...
...failed updating 1 target...
...skipped 3 targets...
...updated 10 targets...
Build finished at 11:12:39.03
Elapsed time: 0:0:31

C:\Users\ghassett\pwiz>
```

chambm, 2019-10-24:
Ah, sorry, I have changed on my VS2019 working copy that haven't been checked in. Here's a patch that should fix it.

[vs2019fix.diff.txt](https://github.com/ProteoWizard/pwiz/files/3768323/vs2019fix.diff.txt)

chambm, 2019-10-24:
Sorry for the hassle. VS2019 isn't officially supported in our build. We build officially with VS2017.

ghassett, 2019-10-24:
Joy!  Thank you so much, Matt!

```
C:\Users\ghassett\pwiz>build-nt-x86\msvc-release-x86_64\msconvert.exe
Usage: msconvert [options] [filemasks]
Convert mass spec data file formats.

Return value: # of failed files.

Options:
  -f [ --filelist ] arg              : specify text file containing filenames
  -o [ --outdir ] arg (=.)           : set output directory ('-' for stdout) [.]
  -c [ --config ] arg                : configuration file (optionName=value)
  --outfile arg                      : Override the name of output file.
  -e [ --ext ] arg                   : set extension for output files [mzML|mzXML|mgf|txt|mz5]
  --mzML                             : write mzML format [default]
  --mzXML                            : write mzXML format
  --mz5                              : write mz5 format
  --mgf                              : write Mascot generic format
  --text                             : write ProteoWizard internal text format
  --ms1                              : write MS1 format
  --cms1                             : write CMS1 format
  --ms2                              : write MS2 format
  --cms2                             : write CMS2 format
  -v [ --verbose ]                   : display detailed progress information
  --64                               : set default binary encoding to 64-bit precision [default]
  --32                               : set default binary encoding to 32-bit precision
  --mz64                             : encode m/z values in 64-bit precision [default]
  --mz32                             : encode m/z values in 32-bit precision
  --inten64                          : encode intensity values in 64-bit precision
  --inten32                          : encode intensity values in 32-bit precision [default]
  --noindex                          : do not write index
  -i [ --contactInfo ] arg           : filename for contact info
(ettc...)
```

ghassett, 2019-10-24:
Moving on with the quest to build Skyline!

Any idea what the correct procedure for building the Skyline Windows GIU app would be?  Do I build that one from within Visual Studio?  I see a number of `.sln` files in the repository but am unsure which one to open (I tried Skyline.sln but it had 18 compilation errors, about missing a DLL (`msparser.dll`) and missing Metadata files e.g. `Skyline-daily.exe` and many others).

chambm, 2019-10-24:
Skyline requires some extra stuff that's optional for the core pwiz build:

## Mascot Parser
Go to http://www.matrixscience.com/msparser_download.html to register with Matrix Science
They'll send you an email, follow the instructions to download v2.6 for windows, both 32 and 64 bit.  You need at least the VS2015 folder
After downloading,and run each 7zip .exe as Administrator.
Run (as administrator) the 32 bit self-extractor, extracting to "C:\Program Files (x86)\Matrix Science\Mascot Parser\"
Run (as administrator) the 64 bit self-extractor, extracting to "C:\Program Files\Matrix Science\Mascot Parser\"

## Checking out Skyline (as part of ProteoWizard)
Note: for performance reasons relating to Sophos Antivirus constantly scanning the stuff you're trying to build, you should create a top-level directory for all your development work and add a Sophos exclusion (you can do this via the Sophos icon in the Windows taskbar).  For example, I (BSP) set mine as "I:\proj" (my machine arrived with a large drive "I:").

## To Build Skyline and ProteoWizard
This step builds the ProteoWizard core library used by Skyline to read mass spec data, as well as Skyline itself, and runs tests on both.  Once you've done this, as a Skyline developer you'll mostly work in the Visual Studio IDE.  If you need to work on the ProteoWizard core, it's back to the bjam-based build system (though you can follow the debugger from Skyline into the pwiz core, if pwiz is built with the "debug" option). 

In these instructions, <root> refers to that umbrella directory you set up earlier as an antivirus exclusion.

Create the batch files b32.bat or b64.bat in this directory by creating a new file text file with a single line like:
pwiz_tools\build-apps.bat 64 --i-agree-to-the-vendor-licenses %*

Where the first argument to build-apps.bat is either 32 or 64.  This is an either/or thing, and needs to agree with your plans to do 32 or 64 bit development in the VS IDE.  If you want to switch from 32 to 64 bit later, you'll need to run clean.bat and rebuild).
  
Save the file as b64.bat or b32.bat and exit.
Type b32.bat or b64.bat in the command-line.
For the quickest possible build without tests type "b64.bat pwiz_tools\Skyline//Skyline.exe"
It will build, then you'll see lots of Skyline windows flashing by as the tests are run.

Testing Note: The test agent "QTAgent" tends to crash, don't be alarmed.  This tends to happen around transitions (debug to release, 64 bit to 32, etc) and doesn't affect the tests - that is, it doesn't crash during any tests, it crashes during test system setup and takedown.

#### BiblioSpec Note : if you want to run all the tests on BiblioSpec, add this to your b64.bat file:  --msparser-path="C:\PROGRA~1\MATRIX~1\MASCOT~1\vs2015"

#### Profiling Note: If you intend to do any dotTrace performance work, you'll probably want to add "debug-symbols=on debug-store=database" to your b??.bat file to make the Proteowizard core more navigable.  if you move to this later you might want to do a clean build to make sure all is consistent.

## Test and Begin Skyline Development
Note that the first time you open Skyline.sln, you are likely to be asked to download and install an update dotNet package - do that.

In Windows Explorer navigate to <root>\proj\Proteowizard\pwiz\pwiz_tools\Skyline
Double-click Skyline.sln
Be sure that your build configuration (x86 or x64) is set correctly.
Make sure your test settings match your build configuration:
Test > Test Settings > Select Test Settings File
and browser for either TestSettings_x86.runsettings or TestSettings_x64.runsettings in the Skyline folder
On the Tools menu, click Options, select Debugging, select General, and uncheck the 'Enable Just My Code (Managed only)' option.
On the ReSharper menu, click Options, under Tools, select Unit Testing, and uncheck the 'Shadow-copy assemblies being tested' option.
In the Test menu, choose Run and click All Tests in Solution
Once all of the tests have passed, you are ready to explore the code, debug and implement new features.

ghassett, 2019-10-24:
Matt, you are awesome.  I cannot thank you enough for your help.  The time you took to help me is genuinely appreciated.  Skyline has been built!

closed 2019-12-03 by chambm