ProteoWizard/pwiz #719: Something was wrong when I build pwiz on ubuntu 16.04

QingzuHe, 2019-11-10:
Hello,
I download source file of pwiz (file name: pwiz-src-3_0_19309_1dc16c9.tar.bz2). When I build on ubuntu 16.04, some problems have occurred.

...failed updating 1 target...
...skipped 1 target...
...updated 1776 targets...
At least one pwiz target failed to build.

Before build pwiz, I install boost library and java open-jdk and git. However, I find strange log information.

================================================================
[MSDataAnalyzerApplication] no files found matching "file0"
[MSDataAnalyzerApplication] no files found matching "file1"
[MSDataAnalyzerApplication] no files found matching "file2"
[MSDataAnalyzerApplication] no files found matching "file3"
[MSDataAnalyzerApplication] no files found matching "file4"
[MSDataAnalyzerApplication] no files found matching "MSDataAnalyzerApplicationTest.temp.txt"
[SpectrumListFactory] Ignoring wrapper: coffee
[SpectrumListFactory] Ignoring wrapper: news media

====== BEGIN OUTPUT ======
no vendor test data found (try running without --incremental)

EXIT STATUS: 1
====== END OUTPUT ======

    LD_LIBRARY_PATH="/usr/bin:/usr/lib:/usr/lib32:/usr/lib64:$LD_LIBRARY_PATH"
export LD_LIBRARY_PATH

    status=0
    if test $status -ne 0 ; then
        echo Skipping test execution due to testing.execute=off
        exit 0
    fi
    name=Reader_Bruker_Test
    echo > /dev/null

[SpectrumList_PeakPicker] Warning: vendor peakPicking requested, but peakPicking is not the first filter.  Since the vendor DLLs can only operate directly on raw data, this filter will likely not have any effect.
Warning: vendor peakPicking was requested, but is unavailable as it depends on Windows DLLs.  Using ProteoWizard centroiding algorithm instead.
High-quality peak-picking can be enabled using the cwt flag.
================================================================

When I use msconvert to convert tdf file to mzML, the flowing message appear on screen: 

Command:
msconvert test.d --mzML --mz32 --inten32 --zlib

Result:
[ReaderFail] [Reader_Bruker::read()] Bruker Analysis reader not implemented: requires CompassXtract which only works on Windows
Error processing file test.d

How can I fix this problem?
Thank you very much!!

chambm, 2019-11-11:
I feel the error message is pretty clear: `Bruker Analysis reader not implemented: requires CompassXtract which only works on Windows`; although if this is a TDF source then the error message is a bit out of date, since there are 3 different Bruker SDKs we use depending the file format. But all pwiz vendor readers actually only work on Windows. Or you can use the relatively new pwiz-skyline Docker container: https://hub.docker.com/r/proteowizard/pwiz-skyline-i-agree-to-the-vendor-licenses

QingzuHe, 2019-11-12:
I got it. Thanks again ^_^

closed 2019-11-27 by chambm