ProteoWizard/pwiz #745: TIC for just MS1

chambm, 2019-11-27:
Since we switched to using pwiz TIC chromatograms for Skyline, the fact that pwiz includes all spectra in the TIC chromatograms has become an issue. Skyline wants a way to get TICs for only the MS1s.

Almost all the important (to Skyline) formats allow getting TIC per ms level:
- Agilent (BDAChromFilter.MsLevelFilter)
- Bruker (TIC is built from SQLite files for BAF and TDF, so it's flexible)
- Sciex (TIC is always per experiment)
- Thermo (using scan filter string)
- Waters MassLynx (TIC is always per function)

The Waters UNIFI reader currently doesn't get chromatograms at all: I'm not sure the API supports it.

I can see a few ways to satisfy Skyline (I've bolded the aspects I find most important):

1. Change the retrieval functions to ignore points from MS levels other than 1 **(could be controlled by a `Reader::Config` option though)**

Pro | Con
------------ | -------------
**Simplest to implement** | TIC arrays sizes will not be equal to number of spectra
the "TIC" id can stay the same | all spectra are not represented (any other users that want a summary of ALL spectra rather than just MS1s will be out of luck)

2. Change the APIs so they return multiple chromatograms, one per MS level. The chromatogram ids would change to "TIC MS1", "TIC MS2" etc.

Pro | Con
------------- | -------------
easiest on downstream code | TIC arrays will not be equal to number of spectra **(although they will be equal on a per MS level basis, so that could be an easy way to count spectra per ms level)**
all spectra are still represented | changing ids may violate assumptions downstream software may have (Skyline certainly expects to find the plain id "TIC" in several places)

3. Change the APIs so they return an MS level array as well as times and intensities, and have Skyline filter the arrays to just the MS1 points.

Pro | Con
------------- | -------------
all spectra are represented | hardest on downstream code
**does not change behavior from previous pwiz versions, only adds a new array** | must use some non-standard array for the ms level array
chromatogram keeps "TIC" id |

I'm leaning toward option 3 because it doesn't change the previous behavior. It'll be totally backward compatible. What do you think @brendanx67 @bspratt ?

chambm, 2019-12-12:
From Nick:
> It looks like we still have not fixed the problem where the Total Ion Current area needs to only include MS1 scans.

>We do need to fix this before we ship 19.2 because a lot of labs normalize against ChromFileInfo.TicArea.

>My guess is that the best thing to do would be to add a value to ReaderConfig which says whether the TIC and BPC should be MS1 only. (option #1)

OK. I'll knock this out.

chambm, 2019-12-12:
@nickshulman @brendanx67 What do you want to see for files that have no MS1s? Chromatogram with 0 time points?

nickshulman, 2019-12-12:
Chromatogram with zero points sounds good.
(I think any behavior would be fine).
-- Nick

closed 2019-12-18 by chambm