ProteoWizard/pwiz #775: Issues converting Waters DDA to mzML or .mgf

Gscorreia89, 2019-12-09:
Hi,

I have been trying to convert some Waters DDA files in .RAW to both .mzML and .mgf but the conversion keeps crashing when opening the file (won't even reach the SHA1 checks).

If I use the msconvert GUI it just crashes without any error message at all, but when trying with the command line I get the cryptic: "caught unknown exception" please report this error to ... .

I have done this successfully in the recent past (over the summer) with similar files, but I can't seem to get anything with the newer msconvert version. 

The files I am trying to convert now are in continuum/profile mode and I also wanted to centroid them directly with ProteoWizard as well, but I get a crash at the file opening stage whatever the parameters and filters I use (whether I use peak detection from vendor or CWT first and no filters or filters after, or without peak detection, export to .mzML or .mfg, etc.). 

The proteowizard version I am using is: 
ProteoWizard release: 3.0.19337 (c0705bc76)
Build date: Dec  3 2019 18:18:00
and I have installed the latest .NET framework (4.8). I got the same error on Windows 10 and Windows 7.

I can share some example .RAW files, just let me which way you prefer me to pack and share them.

Best Wishes,
Goncalo Correia

chambm, 2019-12-09:
Yes I think I'll need to see the raw file to diagnose this. Whatever is the most convenient way for you to transfer it would be fine. If you don't have a way, you can register and upload them here: https://skyline.ms/project/home/support/file%20sharing/begin.view?

Gscorreia89, 2019-12-09:
Dear Matt,

I just sent an example file via email (OneDrive/ outlook sharepoint link) to support@proteowizard.org. Let me know if you have any questions about the file or problems accessing it.

Thank you very much for looking in this.

best Wishes,
Goncalo Correia

chambm, 2019-12-09:
`We're sorry, but matt.chambers42@gmail.com can't be found in the imperiallondon-my.sharepoint.com directory. Please try again later, while we try to automatically fix this for you.`

Gscorreia89, 2019-12-09:
Sorry, forgot that annoying detail  - I have resent using another service which hopefully works! You should be receiving an email soon with a download link.

chambm, 2019-12-09:
Does this file open in Waters MassLynx? It's indeed crashing as soon as it tries to open it in the vendor API.

Gscorreia89, 2019-12-09:
It does! And it opens in mzMine2 and I was able to get an .mgf with mzMine2 :S

chambm, 2019-12-09:
That's funny. They're using a much older version of the vendor DLLs, so something obviously regressed in the meantime. Mind if I forward the raw to Waters for them to fix?

Gscorreia89, 2019-12-09:
Yep, feel free to share with them. Your theory makes sense to me: I did convert some DDA .raw acquired in our group to .mzML and .mgf over the summer and it worked :S, but now none of the files we have which worked in the past are working, so it was working for a while!

chambm, 2019-12-10:
My Waters contact said the file you sent should have 5 functions, not 3. Have you deleted some files? Anyway he agreed it shouldn't crash because of that though, so there will be a fix.

Gscorreia89, 2019-12-10:
?? - I just checked in case there was any problem copying the file etc but everything is as is supposed, and as reported in the ._extern.inf - 3 functions only. 

What would the 5 functions be? I was under the impression that a DDA like this should have MSLevel 1, MSLevel 2 and Lockspray, and not more - is this something that I should look into fixing and can cause problem with these files or its all good for now??

Thank you very much for looking this up so swiftly, including checking with Waters.

chambm, 2019-12-10:
Yeah there does seem to be a mismatch between extern.inf and header.txt. According to header.txt there are 5 functions calibrated. According to extern.inf function 1 is MS2 and function 2 is MS1. I've never seen that.

Gscorreia89, 2019-12-10:
Interesting... I just checked on an instrument with a colleague and I have a theory about this... 

We routinely run experiments with extended dynamic range (DRE), and I think these 5 functions come from calibrating the instrument with DRE option on. But the DDA itself (ran using the FastDDA function) only has 3 functions, and the DRE is not applied in this case. In massLynx we can see also 5 functions, but the last 2 are completely empty ... So there are 3 real and 2 which might just be an artefact from running DDA with DRE pre-configured??? Does this makes sense? 

Please let me know if you need more detail about this.

chambm, 2019-12-11:
Fixed in #792

Gscorreia89, 2020-01-14:
Hi, 

Any update on this? Is the problem fixed in the latest version?

Thanks,
Goncalo

chambm, 2020-02-11:
No there hasn't been a Waters API update, which is what would be needed to fix the crash. I've pinged my contact about it.

chambm, 2020-02-14:
A status update for you: I got new Waters DLLs that fix this but it's not a total fix. Your file still crashes if I try to lockmass correct it. So I added a workaround so that lockmass could be skipped if it causes an error, but I'm not sure it's worth releasing. I was also getting some strange errors on all Waters RAW data after a while of testing the new DLLs. I may come back to it later.

Gscorreia89, 2020-02-18:
Interesting. Please let me know if you need any more data or information. In case it helps to understand the problem and settings we use, feel free to pass my email to your contact at Waters as well. 

Thanks for looking into this,
Goncalo

Gscorreia89, 2020-08-13:
Dear Matt,

We recently ran some DDA experiments to confirm my hunch that this was related to an odd DRE + DDA interaction. The problem is that we were running DRE with diagnostic mode ON (saving to separate functions the low and high intensity data streams in addition to the final processed DRE data), which causes the mismatch you observed in the recorded and expected number of features! I have managed to create files replicating the issue, and even "manually patch" old files so they are convertable to .mzML and .mgf with proteowizard's msconvert. 

Thank you very much (and your contact at Waters) for your tests and information which pointed me in the direction of the root problem! Since this is such an edge case, I am not convinced it would be beneficial to do any modification to the msconvert libraries (and the issue should be closed), but please let me know if you think otherwise and/or would like more information or example files. 

Thanks for looking into it,
Gonçalo Correia

closed 2020-09-16 by chambm