ProteoWizard/pwiz #890: MSConvertGUI.exe

IrisZheLi, 2020-01-30:
I tried to run MSConvertGUI.exe
1) Launched MSConvertGUI.exe under [ pwiz-master\build-nt-x86\msvc-release];
2) Selected example_data and clicked [Start] button.
3) An error message box showed up;
4) I clicked [Continue] button, hangs up...
I have tried diff. sample files, same results.
Please help. thanks; - Iris Li

-) Screen shot:
[image]
-) Details
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.
************** Exception Text **************
System.ArgumentException: Illegal characters in path.
   at System.IO.Path.CheckInvalidPathChars(String path, Boolean checkAdditional)
   at System.IO.Path.GetFileName(String path)
   at System.IO.StreamReader..ctor(String path, Encoding encoding, Boolean detectEncodingFromByteOrderMarks, Int32 bufferSize, Boolean checkHost)
   at System.IO.StreamReader..ctor(String path)
   at MSConvertGUI.MainLogic.ParseCommandLine(String outputFolder, String argv) in C:\SoftwareDevelopment\LC-MS\ProteoWizard\pwiz-master\pwiz_tools\MSConvertGUI\MainLogic.cs:line 337
   at MSConvertGUI.ProgressForm.ProgressForm_Load(Object sender, EventArgs e) in C:\SoftwareDevelopment\LC-MS\ProteoWizard\pwiz-master\pwiz_tools\MSConvertGUI\ProgressForm.cs:line 167
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4075.0 built by: NET48REL1LAST
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
MSConvertGUI
    Assembly Version: 3.0.20006.0
    Win32 Version: 3.0.20006
    CodeBase: file:///C:/pwiz-master/build-nt-x86/msvc-release/MSConvertGUI.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4042.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4001.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.3752.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4110.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
STL_Containers
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/pwiz-master/build-nt-x86/msvc-release/STL_Containers.DLL
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.3752.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.3752.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
CustomDataSourceDialog
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/pwiz-master/build-nt-x86/msvc-release/CustomDataSourceDialog.DLL
----------------------------------------
pwiz_bindings_cli
    Assembly Version: 0.0.0.0
    Win32 Version: 
    CodeBase: file:///C:/pwiz-master/build-nt-x86/msvc-release/pwiz_bindings_cli.DLL
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.3752.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.
For example:
<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>
When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

chambm, 2020-01-31:
Interesting. Did you build this MSConvertGUI yourself from the source code?

IrisZheLi, 2020-02-03:
Yes, I built it using quickbuild.bat --i-agree-to-the-vendor-licenses with no other options.

IrisZheLi, 2020-02-04:
Chambm, thanks for your reply.
See attached  [MSConvertGUI_Build_ReleaseX86.docx](https://github.com/ProteoWizard/pwiz/files/4156521/MSConvertGUI_Build_ReleaseX86.docx)

chambm, 2020-02-04:
Thanks for the thorough screenshots. Have you tried the 64-bit build? That's a strange error. I don't see anything obviously wrong with your settings.

chambm, 2020-02-04:
Nevermind I see the problem. You have "File of filenames" selected. You want "List of files". "File of filenames" is like if you give it a text file where each line is the path to a file you want converted.

chambm, 2020-02-04:
Granted we could definitely be doing a better job of validating that instead of that totally unhelpful error. Thanks for the report. :)

closed 2020-02-04 by chambm

IrisZheLi, 2020-02-05:
build [debug, x64]
[MSConvertGUI_Build_DebugX64.docx](https://github.com/ProteoWizard/pwiz/files/4156983/MSConvertGUI_Build_DebugX64.docx)

IrisZheLi, 2020-02-05:
Chambm: You want "List of files". 
Thanks, it works, see the last page.
[MSConvertGUI_Build_ReleaseX86.docx](https://github.com/ProteoWizard/pwiz/files/4157101/MSConvertGUI_Build_ReleaseX86.docx)