ProteoWizard/pwiz #915: Repeated mz values

griquelme, 2020-02-10:
Hello, I am using msconvert to convert raw files to mzML with centroiding (using cwt filter with SNR=1 and a minimum peak spacing=0.01), but after converting the files I noticed that some of the scans have repeated mz values.

I am using MSConvert 3.019269 1e153dfa2 on Windows 10 Enterprise 2016 LTSB, but I also saw this bug on Windows 7 SP1.
The files I am trying to convert are Waters RAW files, but I also saw repeated values on Thermo Files.

chambm, 2020-02-10:
Can you give more details and examples? Is it Waters IMS data?

griquelme, 2020-02-11:
It's LC-MS data from a Waters Xevo G2-S QTof. I can provide a .RAW file as an example if you want.
Duplicate values only appear when performing peak picking on profile data. If no centroiding is done there aren't any problems...
I noticed the duplicate values when I was processing the converted data using pyopenms, but I don't think it's a problem of pyopenms

chambm, 2020-02-11:
Yes I'll need an example file.

griquelme, 2020-02-12:
Hi, sorry for the delay. [Here](https://drive.google.com/file/d/1upxreaZZlv2Ng9CqMZ0Gf51xdOXVaMTZ/view?usp=sharing) is a link for one of the RAW files were I detected duplicated values. The filters I used on msconvert were:

1. Subset with MSLevel = 1
2. cwt with SNR=1 and minimum peak distance=0.01

in [this](https://gist.github.com/griquelme/8a16a91d6659261f5587e30855a8e99d) python script you can see where the repeated values appear:

chambm, 2020-02-18:
Fixed in #937 . Only for CWT though. I noticed that the Waters peak picker also returns duplicate peaks. I may fix that in a separate PR, but for now it is returning the peaks exactly as Waters provides them. I also added a centroid option to CWT (defaults to off for now) which will use the centroid (weighted average) of each peak profile as the m/z rather than the apex m/z. I didn't make it the default because I haven't tested it enough. But it should theroetically be good to use on Waters IMS data after --combineIonMobilitySpectra and scan summing, because in that case their native centroiding doesn't work.

closed 2020-02-18 by chambm