ProteoWizard/pwiz #959: Error during bruker file conversion (BAF) to MzML with MSConvert (version 3.20058)

bjonnh, 2020-02-27:
There were other files acquired with the same method that worked. Just a few files failed. They work properly in DataAnalysis. I can send that specific sample data if needed (~700M).

This is what the log says:

Starting...
Opening file "E:\sjin28\20200213-3502-A2-microfrac-redownload\040_1-45_1_274.d" for read...
Failed - System.Exception: File format problem: expected different segment.
   at pwiz.CLI.msdata.ReaderList.read(String filename, MSDataList results, ReaderConfig config)
   at MSConvertGUI.MainLogic.processFile(String filename, Config config, ReaderList readers, Map`2 usedOutputFilenames)
   at MSConvertGUI.MainLogic.Go(Config config, Map`2 usedOutputFilenames)

chambm, 2020-02-27:
Yeah, I'll need to see a failing file to understand this. Are these native BAF files or converted from FID?

bjonnh, 2020-02-27:
native BAF.
Do you have a place where I can post that?

chambm, 2020-02-27:
Google drive would be fine.

bjonnh, 2020-02-27:
Shared to your …42@gm… email.

bjonnh, 2020-02-27:
I see the same error when using baf2sql: 
BAF file = 040_1-45_1_274.d/analysis.baf
(Re-)Generating SQLite cache...
2020-02-27 15:46:26.933307 [tid=0xe15fa7c0] [INFO ] bdal.io.baf2sql: Generating new SQLite cache in analysis directory: 040_1-45_1_274.d/analysis.sqlite
*** caught std::exception: ***
what() = File format problem: expected different segment.
boost::diagnostic_information() = /home/jenkins/workspace/de.bdal.cpp.io.baf2sql-c-linux-x86_64-gcc6-parameterized/src/include/io/baf2sql/baf2sql_cpp.h(38): Throw in function void baf2sql::throwLastBaf2SqlError()
Dynamic exception type: boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<std::runtime_error> >
std::exception::what: File format problem: expected different segment.

bjonnh, 2020-02-27:
Do you have a contact at Bruker that work on baf2sql?

chambm, 2020-02-27:
Can you also share one of the runs from the same method that doesn't have the problem? I agree it's clearly in the Bruker API so it's best to forward to them.

bjonnh, 2020-02-27:
Sure, let me push that in the same directory.

bjonnh, 2020-02-27:
I renamed them with WORKING/NOTWORKING

chambm, 2020-02-28:
Do you know where the BackgroundLinePos.ami and DensViewPos.ami files came from? I think those are calibration files.

bjonnh, 2020-02-29:
They are likely coming from Data Analysis. We do a manual calibration step with an automated method in DataAnalysis (internal calibration). We do that so our files are calibrated, at least that's how baf2sql does it, I hope ProteoWizard keep the calibration, I will have to check that.

chambm, 2020-03-02:
So why doesn't the "non-working" file have those calibration files? Does it process properly (not just open) in DataAnalysis?

bjonnh, 2020-03-02:
It works well in DataAnalysis, it just fails in BAF2SQL and ProteoWizard.

bjonnh, 2020-03-02:
I will check if all scans are accessible in DataAnalysis, there may have been a file corruption, but it happened on quite a few files and that's the first time we see that.

chambm, 2020-06-03:
Hi @bjonnh , sorry it took a while to get this to my Bruker contact and back. He says:

> It seems like your customer came across a bug in the acquisition software used (otofControl 5.2.0.7). Inspection of the method shows that there are two pretty short time segments defined in the method (unit is minutes):
> 
```
<segment endtime="0.01"> ...
<segment endtime="0.0316667"> ...
<segment endtime="0.223333">...
<segment endtime="-35.525">...
```
> 
> What can happen in such a case is that due to timing variations (especially in the case of low rate of spectrum acquisition), not a single measured spectrum falls into one of those short segments, in which case the data-writing layer in otofControl is confused and writes broken data.
> 
> I suggest your customer fix the method by removing the first two segments, or at least making them long enough (i.e., much longer than the time it takes to acquire a spectrum).

You said DataAnalysis is able to handle this data, which indicates the "broken data" can be handled more gracefully, but he also said:

> There may be ways to fix the existing data and / or add a workaround to Baf2Sql, but I’m not sure I have capacities for that right now.