ProteoWizard/pwiz #963: Cannot calculate spectra purity with msconvert 3.0.20059

prvst, 2020-02-28:
Hi,

We recently noticed that our [Philosopher](https://philosopher.nesvilab.org/) pipeline is not able to calculate spectra purity for data sets labeled with TMT. We apply a method that is very similar to what P.D. does, but they call it Isolation Interference. The issue seems to only happen when the Thermo raw file is converted using msconvert 3.0.20059, the problem cannot be reproduced with previous versions. What we expect from the method is to have a score that goes from 0 to 100% purity for each spectra (or isolation interference), but with the latest msconvert version every spectra is either 0 or 1. I inspected the mzML files and I don't see any major differences with one exception, see below:

[image]

Apparently the latest version has an extra `<userParam>` tag inside the MS2 scans, right below the `<precursor>` tag. I don't know if that is the issue because my parser was able to read the spectra.

Regarding the purity score issue, everything else seems to be OK, including our TMT quantification, the values from the newest and the previous version seem to match.

I would appreciate if you could help us figuring out this.

Thanks

chambm, 2020-02-28:
I can't help much without knowing what metadata the Philosopher purity calculation is using in its input. You can give me 2 mzML files, one working, one not, and I can tell you the differences between them. If the userParam ms level is the only difference then that's clearly a bug in Philosopher. (Although I suspect precursor intensity might have changed too because not too long ago we changed the way we were calculating that.)

But if you're the developer of that algorithm (or at least can debug the source code), you're in the best position to understand why it's only producing 0 or 1. Let me know if you notice something wrong/different with the mzML output which I could fix/explain.

prvst, 2020-02-28:
The pipeline does not convert the files, we convert them manually and I did so using the Windows GUI. For each PSM we look at the MS1 scan where the precursor peak is and we then calculate the purity by looking around that peak for other peaks.

We are using the same version for comparing things, same processing steps, same database and same parameters, the only difference is the mzML file created by msconvert GUI 3.0.19 and 3.0.2.

I uploaded the files [here](https://umich.box.com/s/4o4bdrso75srvw33c39gkdh94l5hb5d1) for you to take a look.

chambm, 2020-02-28:
For the most part, the userParam ms level and pwiz version are the only differences. However, the index on the newer file is corrupt after index 4121 because 4121 is missing the userParam ms level. I'd guess that's because the file was edited manually to remove it. Perhaps you did that while trying to test it? But if you didn't edit the file, I'll need the RAW file to try to reproduce this weird artifact.

Having a partially corrupt index actually led me to some other index handling bugs in pwiz, so that was a good thing to catch in any case. :)

prvst, 2020-02-28:
Yes you are right, I removed to test, sorry for not noticing before sending to you.

prvst, 2020-02-28:
Actually, let me clarify, I only removed that `<userParam>` that I highlighted above

chambm, 2020-02-28:
OK in that case it's probably a bug in Philospher's mzML reader. It should just ignore those userParams and not let it affect the calculation.

chambm, 2020-02-28:
To be clear, the binary data is identical between the two files.

prvst, 2020-02-28:
Yes, that's why I was suspecting that the difference could be on the meta data or file structure

prvst, 2020-02-28:
But if that's a bug on our reader, how come the previous versions of msconvert work without any issue? That tag is not present when converting using 3.0.19

prvst, 2020-02-28:
BTW, we also tested on the same file but converted with version 3.0.7 and it also works fine but we can't find that version to download.

chambm, 2020-02-28:
The previous versions didn't have that userParam. Zero or more userParams are allowed in that element. It should not affect your reader, but it's pretty clear that it is, right?

chambm, 2020-02-28:
I'm not sure what versions you're quoting. The versions for those files are 3.0.20059 (newer) and 3.0.19127.

prvst, 2020-02-28:
The file that can be found [here](https://wetransfer.com/downloads/348bfd53779d1a861b2a7301df31a10d20200225135412/7d4bf2) says the following:

```
      <software id="pwiz" version="3.0.7680">
        <cvParam cvRef="MS" accession="MS:1000615" name="ProteoWizard software" value=""/>
      </software>
```

chambm, 2020-02-28:
That's a pre-git version. But I'm not sure how that's related to the fact that the userParams are there in the mzML now, and they do not make the mzML invalid. Is it hard to fix your mzML reader so it ignores it?

prvst, 2020-02-28:
not at all, I'm working on something now but at the same time wanted to confirm with you that this was the only difference. I'll try that and I'll let you know. Thanks for replying

closed 2020-02-28 by prvst