ProteoWizard/pwiz #1034: issues with shimadzu lcd spectra

banterle, 2020-04-29:
I have a few Shimadzu spectra (.lcd files), the program stops with an exception reporting that the file has a wrong year, which is not in the range 1400 - 9999.

chambm, 2020-04-30:
A workaround for this was added last November. It should clamp the date to always be greater than 1400 and less than 10000.

banterle, 2020-04-30:
Thank you for your update. I downloaded the latest compiled version of msconverter last week. Is it updated? Do I need to download pwiz and compile msconverter?

chambm, 2020-04-30:
This fix was put in last November. What version do you see in the help/about text?

banterle, 2020-04-30:
I used the version 3.0.20121.1edcd2790-x86_64

chambm, 2020-04-30:
Can you share the lcd file?

banterle, 2020-05-01:
I can share with you the LCD file by e-mail, but I need to ask you not to share it online if it is possible. Does it sound reasonable?

banterle, 2020-05-05:
any luck?

chambm, 2020-05-05:
Which program are you getting the error in? Is it with this lcd file? I don't see any chromatograms or spectra in this file.

banterle, 2020-05-05:
I used MSconverter with this file, and it should have spectra in it. 

I got a similar errore (year) even with this one:
www.banterle.com/francesco/work/spectra/chestnut.lcd

chambm, 2020-05-05:
Is this from an IT-TOF?

banterle, 2020-05-05:
Nope, it's from HPLC-DAD Shimadzu (liquid chromatography).

chambm, 2020-05-05:
Hmm. No MS in there? I know IT-TOF is **not** supported, but the APIs we've been given are specifically for QQQ and QTOFs, so it's likely that **only** those instruments are supported. Specifically, using Shimadzu's own API testing suite, I can't find any data in your file.

banterle, 2020-05-05:
No MS in there. There should be some spectra in there, that's strange.

closed 2020-05-11 by chambm