ProteoWizard/pwiz #1088: Source only repo?

hojalot, 2020-05-22:
I started to clone this repository, but stopped less than half way through after realizing it is over 1GB.  I wonder if it's due to multiple versions of binaries stored within.

Is there a smaller source-only version of this repository?

chambm, 2020-05-22:
Try cloning with depth=1 so you only get the latest version of the binaries. But yeah, that's the biggest reason for the overall repo size. Probably ought to clean that out some day.

chambm, 2020-05-22:
Oh, and there are source tarballs available at http://proteowizard.sourceforge.net/download.html (Source, bjam build, with or without vendor reader support)

closed 2020-05-26 by chambm

hojalot, 2020-05-28:
Using depth=1 worked fine.  Thanks for the suggestion!