ProteoWizard/pwiz #1127: Thermo raw file conversion to mzXML and mzML missing all MS1 scans

ariskay, 2020-06-25:
Hello,

We are having issues converting Thermo raw files to mzXML and mzML. The resulting files are missing all the MS1 scans. This is not happening with older raw files. As a test, we decided to convert the raw files using reAdW, and the resulting file includes the MS1 scans. On the other hand, we reviewed the raw file with the Xcalibur software, and the scans are there.

Here the version of msconvert extracted from the converted mzXML:

```
    <dataProcessing>
      <software type="conversion" name="ProteoWizard software" version="3.0.20083"/>
      <processingOperation name="Conversion to mzML"/>
    </dataProcessing>
```

I would be glad to provide further information in order to reproduce the issue,

Thank you

chambm, 2020-06-25:
Are the MS1 scans "SIM" scans (SIM would be in the filter)?

ariskay, 2020-06-25:
Checking the one generated by reAdW the MS1 scans are SIM:

 ```  filterLine="FTMS + c NSI SIM ms [394.9049-562.9049]"```

Thanks for the clue, I checked the SIM as spectra check in the MSConverGUI and now the MS1 scans are showing up :relaxed:

chambm, 2020-06-25:
Indeed Thermo has been calling not-really-SIM scans SIM for a while now. I'd like to have a heuristic to decide between true SIM (maybe < 10m/z wide) and "Thermo SIM", but haven't really settled on definitive criteria.

closed 2020-06-25 by chambm