ProteoWizard/pwiz #1146: Failed to build idpicker in Microsoft Visual Studio 2017

IrisZheLi, 2020-07-12:
I successfully ran the build command:
`quickbuild.bat --i-agree-to-the-vendor-licenses address-model=64`

I am attempting to build the proteowizard project in Microsoft Visual Studio 2017.
I tried IDPicker; \..\pwiz_tools/Bumbershoot/idpicker/IDPicker.sln
-) STL_Containers
-) CommonUtil
-) ZedGraph
-) MSGrapha
-) Uitl
-) Seems 
Passed;

But it failed when building Model ;  
in idpicker\Model\Settings.cs which complains that 'Qonverter' could not be found
line 64 -- public virtual Qonverter.QonverterMethod QonverterMethod { get; set; }
In References, it missed InteropQonverter.

Does anyone know the solution for this?
Thanks;

chambm, 2020-07-13:
The default quickbuild.bat command probably doesn't build Bumbershoot. Try adding `pwiz_tools/Bumbershoot/idpicker//install` to the quickbuild command. That should let you build and run IDPicker from Visual Studio.

IrisZheLi, 2020-07-13:
Matt :
Thanks for your quick reply.Try adding
pwiz_tools/Bumbershoot/idpicker//install
I tried this:
C:\pwiz>quickbuild.bat --i-agree-to-the-vendor-licenses address-model=64
pwiz_tools/Bumbershoot/idpicker//install
But failed.
Mayme my command is not right;

Iris Li

chambm, 2020-07-13:
I need more detail that "But failed." :)

IrisZheLi, 2020-07-13:
Matt:
I input this, which failed;
quickbuild.bat --i-agree-to-the-vendor-licenses address-model=64
pwiz_tools/Bumbershoot/idpicker//install
Could you please do me a favor, what is the command to build a debug
version?  I like to try again.

I am not good at batch files, I like to use *.sln in VS;
I have built Skyline and Topograph in VC, x64 debug in VC successfully and
MSConvertGUI with your help.
Thanks;
Iris Li

chambm, 2020-07-13:
All the pwiz tools must be built from command-line first to get the C++ bindings built. The Visual Studio projects that actually build applications (IDPicker, Skyline, Topograph, MSConvertGUI, SeeMS) don't know anything about the C++ core, they only use the C++/CLI bindings via C#.

What I need instead of "failed" is the output of the command, telling me what the error is.