ProteoWizard/pwiz #1149: Conversion from Bruker MS Imaging .d format (mcf, mcf_idx)

fgrelard, 2020-07-21:
Currently, it is not possible to convert Mass Spectrometry Imaging data obtained directly from a Bruker MALDI-TOF instrument. First, we have to convert it to multiple `fid` subdirectories through FlexImaging, and then use `msconvert` to get mzML files and eventually an imzML image. 

The original exported format contains MCF and MCF_IDX files. I tried searching for documentation online but came up empty. 
One example can be found online => [here](https://doi.ipk-gatersleben.de/DOI/269b0f6b-2bf9-4d31-b6b0-70639a8416a2/6e0a9d5c-ebcc-4802-8411-26de6f4cc281/0/1847940088).

Is the export from this format to mzML something that you considered adding to pwiz?

chambm, 2020-10-26:
The example you linked to above does have a 1.2GB analysis.baf file. That's where the data is. The MCF files are some kind of metadata.

fgrelard, 2020-10-27:
An imaging example is available [here](https://zenodo.org/record/3476207). The data is stored in MCF files.
See `Kidney_30min` -> `.d` subfolder.

chambm, 2020-10-27:
Those datasets do have some big mcf files, but I'm not totally convinced that's where the data is. Looking at them in a hex editor I can't discern any obvious data arrays of floats or doubles.

fgrelard, 2020-10-29:
Thanks for looking at this issue.
I will check whether this data is sufficient for the full conversion to XMASS `fid` through the proprietary software. 
Unfortunately, I won't have access to the software in the near future. I will get back to you as soon as I can.

chambm, 2021-01-21:
I didn't mean to imply the .d directory referenced files anywhere else. Only that some files might be missing from it. If you find any SDK that can read this data, we can integrate it to read the files directly, but your conversion using Bruker's converter to fid (BAF would be better if you can do that) may be your only option now.

hhh-123, 2022-07-16:
Hello, I am a little bit confused about the solution.
1. what's the meaning of analysis.baf file or what's the analysis.baf file representing for?
2. what do the word "fid" and the word "BAF" mean?
Thanks!