ProteoWizard/pwiz #1150: msdata couldn't read the converted mzML from wiff file with latest version of msconvert (both docker and windows gui)

Danissss, 2020-07-21:
Docker version:
```
docker pull chambm/pwiz-skyline-i-agree-to-the-vendor-licenses:latest
```

Windows GUI version:
```
3.0.20200-b27502862 (developer build)
```

The issue from msdata  `readSRMData` is here:
```
Error: Can not open file /converted_20000615-1031350968_1_1_16_20000615.mzML! Original error was: Error in pwizModule$open(filename): [IO::HandlerBinaryDataArray] Unknown binary data type.
```

The previous Windows GUI version works for wiff convert and `readSRMData` (with exact same parameters):
```
3.0.20125-21f4a60fb (developer build)
```

chambm, 2020-07-21:
Try the Docker image now. The mzR issue is known: https://github.com/lgatto/MSnbase/issues/517

Danissss, 2020-07-21:
Hi, thanks for the reply.

I did try the latest docker image,

the commands that I used are following:

```
sudo docker run -it --rm -e WINEDEBUG=-all -v ~/analyst/wiff/:/data chambm/pwiz-skyline-i-agree-to-the-vendor-licenses:latest wine msconvert /data/*.wiff --mzML  --outdir "mzML/"

sudo docker run -it --rm -e WINEDEBUG=-all -v ~/analyst/wiff/:/data chambm/pwiz-skyline-i-agree-to-the-vendor-licenses:latest wine msconvert /data/*.wiff --mzML -z --64 --filter "msLevel -2" --outdir "mzML/"

sudo docker run -it --rm -e WINEDEBUG=-all -v ~/analyst/wiff/:/data chambm/pwiz-skyline-i-agree-to-the-vendor-licenses:latest wine msconvert /data/*.wiff --mzML -z --64 --filter "msLevel 2" --outdir "mzML/"
```
None of them works for me with ` readSRMData`

My machine info: 

```
Mac Catalina 10.15.5 (19F101)

Docker 2.0.0.3

R: ‘4.0.2’. 
other attached packages:
 [1] msdata_0.28.0       xcms_3.10.1         MSnbase_2.14.2      ProtGenerics_1.20.0
 [5] S4Vectors_0.26.1    mzR_2.22.0          Rcpp_1.0.5          BiocParallel_1.22.0
 [9] Biobase_2.48.0      BiocGenerics_0.34.0
```

I can send you a sample wiff if you want.

Thank you!

chambm, 2020-07-21:
I thought you meant the latest docker image can't open the file. I linked to the issue about readSRMData. They need to upgrade their pwiz version.

closed 2020-07-27 by chambm

Danissss, 2020-09-18:
Hi @chambm , is that possible to have old docker file that can generate compatible mzML to current readSRMData?

chambm, 2020-09-18:
There are older tags on DockerHub. Something from 2019 (3.0.19xxx) should work.