ProteoWizard/pwiz #1152: msconvert asks for display access for some files

truell20, 2020-07-21:
I'm running msconvert like so: `msconvert file.wiff --gzip -o dir/ --filter "peakPicking true 1-",`

For some of my wiff files, I get the following exception:

```[Reader_ABI::read()] unhandled exception

       Unhandled Exception: System.NullReferenceException: Object reference not set to an instance of an object.

                                                                                                                   at Clearcore2.Data.Common.Sample.PF.MassSpectrometerSampleImplementation.Dispose(Boolean disposing)

             at Clearcore2.Data.Common.Sample.PF.MassSpectrometerSampleImplementation.Finalize()

                                                                                                wine: Unhandled page fault on read access to 0x00000000 at address 0x1f9f7316 (thread 003b), starting debugger...

     Can't attach process 0038: error 5

                                       0046:err:winediag:nodrv_CreateWindow Application tried to create a window, but no driver could be loaded.

                                                                                                                                                0046:err:winediag:nodrv_CreateWindow Make sure that your X server is running and that $DISPLAY is set correctly.```

chambm, 2020-10-26:
It's probably trying to create an error window, perhaps a Windows crash dialog. Or does the file finish converting just fine?

sneumann, 2022-03-14:
Hi Michael, has that issue gone away ? If yes, feel free to close the issue. 
If not, could you be a bit more specific ? Which operating system, which pwiz version ? 
Yours, Steffen