ProteoWizard/pwiz #1160: Conversion Failure for mixed orbitrap (MS1) / linear ion trap (MS2) HCD Thermo ID-X

PMSeitzer, 2020-07-28:
proteowizard version:
`Version: 3.0.20034-d7e41e31f (developer build)`

To reproduce:
generate a file on Thermo ID-X, with the following settings:

MS1: orbitrap
MS2: linear ion trap with HCD fragmentation
this also does MS1 SIM scans.

see screenshot:

[image]

Please inquire data transfer.  If anyone has any ideas, I am happy to take this issue on.

PMSeitzer, 2020-07-28:
a likely related error message is produced when running with

MS1 orbitrap
MS2 ion trap with CID fragmentation (instead of HCD as in above comment):

[image]

A run of only the MS2 linear ion trap with CID fragmentation (no MS1) converts successfully.
A run of only the MS2 linear ion trap with HCD fragmentation (no MS1) does not complete successfully.

chambm, 2020-07-28:
Thanks for the report. Please share one of the problematic files for us to fix. Definitely seems like a bug.

PMSeitzer, 2020-07-28:
Thanks @chambm, Here is a link to two files that produced slightly different errors, with descriptions:

https://drive.google.com/drive/folders/1tpD5XgcUdi74YbABKXWf3HN99uTRwyMr?usp=sharing

chambm, 2020-07-30:
What version of pwiz are you using? I have to use the `--ignoreUnknownInstrumentError` to get it to convert at all, and then the "no matching instrument configuration" is more of a warning than an error. Nevertheless this ID-X instrument needs to be added so a fix is required.

PMSeitzer, 2020-07-30:
Hi Matt,

Here is the proteowizard version information:
[image]

I tried to convert the file using the `--ignoreUnknownInstrumentError` flag. However, the conversion was unsuccessful (see screenshot):

[image]

chambm, 2020-07-30:
Can you try the current version? I get:
[image]

chambm, 2020-07-30:
Fixed in https://github.com/ProteoWizard/pwiz/pull/1161 (new instruments added; Orbitrap ID-X required a CV update)

closed 2020-07-30 by chambm

PMSeitzer, 2020-07-31:
Thanks @chambm, I can confirm that the new version works when I use the `--ignoreUnknownInstrumentError` flag.