ProteoWizard/pwiz #1162: Conversion failure centroiding MS3 spectra for ID-X instrument

PMSeitzer, 2020-07-31:
This is a follow-up case, using same setup as #1160 and #1161 

In this case, I collected profile mode MS1 / MS2 / MS3 data on an ID-X, and saw the following error message:
[image]

I noticed that when I only centroided the MS1/ MS2 data, I did not see the error.:

[image]

[image]

This leads me to suspect that the error is associated to MS3 spectra only.

PMSeitzer, 2020-07-31:
I did not include the screenshot, but when I ran `peakPicking true 1-` instead of `peakPicking true 3-`, I saw the same error as shown in the first screenshot

PMSeitzer, 2020-07-31:
Here is a link to the raw file referenced in the above CLIs:
https://drive.google.com/drive/folders/1FEUs_ugiACAjicIWct-cNfT5XVHCYkBK?usp=sharing

chambm, 2020-07-31:
Hmm. `1-2` will centroid only MS1 and MS2 so should avoid that error. But I will fix it.

chambm, 2020-07-31:
I think that's a corrupt scan (3313). It shows as empty in QualBrowser but the TIC/BPI are similar with adjacent scans that are not empty. Try `--filter "peakPicking true 1-" --filter "index 0-3311,3313-"`

PMSeitzer, 2020-08-04:
Hey @chambm, thank you, we are investigating on our end.