ProteoWizard/pwiz #1166: Spectra missing zero intensity points from Waters Cyclic IM-MS

mobiusklein, 2020-08-01:
Converting a .raw directory from a Waters Cyclic IM-MS experiment to mzML with `msconvert` from ProteoWizard 3.0.19014 produced IM-MS spectra which are missing data points which have an intensity of zero. These spectra are reported as profile, but look similar to centroided:
[image]
[image]

When I use the MassLynx API to read the same spectrum (same function, same retention time and same drift time), the spectrum looks like a regular profile spectrum:
[image]
[image]

If I put a zero intensity point 0.01 m/z between every non-zero intensity point with a gap greater than 0.01 m/z between it and the next non-zero intensity point, I recover something very close to the MassLynx API spectrum:
[image]
[image]

Spectra converted by ProteoWizard which do not have a drift time appear correct:
[image]
[image]

I'm using MassLynx SDK 4.7 (via the Python bindings) for this test, using the `readDriftScan` C API call to retrieve the m/z and intensity arrays, which is used by the `(int&, int&, int&, vector<float>&, vector<float>&)` overload of `ReadScan` in the C++ API ProteoWizard uses for IM-MS scans. If the MassLynx version in the `software` element of the mzML is accurate, it's using version 4.1. Any idea why this might be happening?

I can make the .raw directory available if needed for debugging.

Thank you

chambm, 2020-10-26:
What msconvert options did you use when converting?

mobiusklein, 2020-10-26:
This was a while ago, but I think just the default title maker filter was used. I can upgrade to the latest release and convert a subset of a file to see if it changed?

chambm, 2020-10-26:
I don't think our Waters drift scans have ever had the flanking zero points. The MassLynxRaw.dll says it's version 4.6 (4.1 is probably an old hardcoded string in pwiz).

mobiusklein, 2020-10-26:
I don't understand why us calling the same C library function produces different results then.

It also means I can't use my signal processing tools on this data in mzML format without adding the flanking zeros back in which admittedly is only a minor inconvenience on my part.

chambm, 2020-10-27:
There's actually an msconvert filter for that:
```
msconvert.exe --help-filter zeroSamples

zeroSamples <mode> [<MS_levels>]
This filter deals with zero values in spectra - either removing them, or adding them where they are missing.
  <mode> is either removeExtra or addMissing[=<flankingZeroCount>] .
  <MS_levels> is optional, when provided (as an int_set) the filter is applied only to spectra with those MS levels.
When <mode> is "removeExtra", consecutive zero intensity peaks are removed from spectra.  For example, a peak list
  "100.1,1000 100.2,0 100.3,0 100.4,0 100.5,0 100.6,1030"
would become
  "100.1,1000 100.2,0 100.5,0 100.6,1030"
and a peak list
  "100.1,0 100.2,0 100.3,0 100.4,0 100.5,0 100.6,1030 100.7,0 100.8,1020 100.9,0 101.0,0"
would become
  "100.5,0 100.6,1030 100.7,0 100.8,1020 100.9,0"
When <mode> is "addMissing", each spectrum's sample rate is automatically determined (the rate can change but only
 gradually) and flanking zeros are inserted around non-zero data points.  The optional [=<flankingZeroCount>] value can be
 used to limit the number of flanking zeros, otherwise the spectrum is completely populated between nonzero points. For
 example, to make sure spectra have at least 5 flanking zeros around runs on nonzero points, use filter "addMissing=5".
```

mobiusklein, 2020-10-27:
I have a routine to do it too, but neither is actually guaranteed to give us the same result that the Waters SDK is supposed to be extracting for us.

On one side, if we know we have to add the zeros, why store them in the first place, these files are already atrociously large to begin with, on the other, it makes the data less portable and self-describing.

chambm, 2020-11-02:
If you have a signal processing algorithm that relies on a certain constant sample rate (like Savitzky Golay and CWT) then you almost certainly have to regrid the data points anyway, and if your algorithm doesn't need the constant sample rate, then the slightly different X values for zeros probably won't affect the result.

Changing signal processing based on whether the data is coming in from mzML or from vendor format seems unavoidable to me, because there could have been many other things done to the mzML: thresholding, peak picking, deisotoping, etc.

mobiusklein, 2020-11-02:
My peak picking strategy doesn't rely on a constant sampling rate, but it does assume multiple points per peak are present. The issue is that I can detect when those things are done by looking at the mzML metadata. It's true that if a thresholding filter were applied to the mzML prior to my receiving it _and_ the thresholding filter dropped zero intensity data points, I'd need to do this still. Otherwise, these spectra claim to be profile mode but have all the characteristics of centroid mode data.

That said, my main concern is that the mzML file produced by using ProteoWizard to call the vendor library functions is different from calling the same vendor library functions myself.

chambm, 2020-11-02:
It's true those actions I mentioned should be explicit in the mzML metadata, but handling those is logically similar to handling Waters not-quite-profile data (i.e. IF NOT RAW AND Waters instrument AND profile THEN add zeros). 

However it seems that I have a 4.7 version of MassLynxRaw which does provide the zeros. The 4.6 version doesn't. IIRC we didn't update to the 4.7 version because of a memory leak but I'd have to go back and check my emails.

chambm, 2021-01-21:
OK this issue should be fixed. Also fixed CWT peak picking so it works well with Waters combined IMS spectra.

closed 2021-01-21 by chambm

mobiusklein, 2021-01-21:
A build with the fixed version is available here, correct?
https://teamcity.labkey.org/viewLog.html?buildId=1235894&buildTypeId=bt83&tab=buildResultsDiv

chambm, 2021-01-21:
Yes you could get it from the PR build, but it'll be available from the main download page when this finishes:
https://teamcity.labkey.org/viewLog.html?buildId=1235930&tab=buildResultsDiv&buildTypeId=bt83