ProteoWizard/pwiz #1204: The diaUmpire filter in msconvert.exe does not work for me

MarcIsak, 2020-08-31:
Hi, 

I tried to use msconvert.exe (from Proteowizard 3.0.20239) in Windows 10, along with the diaUmpire params filter. However, the process quits quickly as the following message appears: 

[diaUmpire] params filepath is required (params=path/to/diaumpire.params)`

Even though I provide the file path to the params file, it still keeps asking for the file path... I dont really understand why this is the case. 

 I would be happy if you could help me out on this one! See attached code :). 

Best,

Marc

`C:\Program Files\ProteoWizard\ProteoWizard 3.0.20239.db48a3d0d>msconvert.exe D:\MS_files\MSLibrarian_DIA_test\CK_P1911_228.mzXML --filter diaUmpire "C:\DIA-Umpire_v2_0\diaumpire_se_marc_edit.params" -o D:\MS_files\MSLibrarian_DIA_test
format: mzML
    m/z: Compression-None, 64-bit
    intensity: Compression-None, 32-bit
    rt: Compression-None, 64-bit
ByteOrder_LittleEndian
 indexed="true"
outputPath: D:\MS_files\MSLibrarian_DIA_test
extension: .mzML
contactFilename:
runIndexSet:

spectrum list filters:
  diaUmpire

chromatogram list filters:

filenames:
  D:\MS_files\MSLibrarian_DIA_test\CK_P1911_228.mzXML
  C:\DIA-Umpire_v2_0\diaumpire_se_marc_edit.params

processing file: D:\MS_files\MSLibrarian_DIA_test\CK_P1911_228.mzXML
calculating source file checksums
[diaUmpire] params filepath is required (params=path/to/diaumpire.params)`

closed 2020-08-31 by MarcIsak

chambm, 2020-08-31:
Did you figure out that the params should be inside the filter argument? `--filter "diaUmpire params=path/to/params"`?

MarcIsak, 2020-09-02:
Hi, yes I figured it out at last :). It works nice

jspaezp, 2023-05-31:
Hello!

Sorry for resurrecting the issue but I was wondering where the documentation for the diaUmpire filter is located. I cannot seem to find it in the online filter documentation (https://proteowizard.sourceforge.io/tools/filters.html).

Thanks beforehand!
-Sebastian