ProteoWizard/pwiz #1213: Error building bjam

JB91451, 2020-09-03:
Dear all,

I tried to build pwiz in order to install BiblioSpec on a 64bit PC with win7 running.
When running the comand (cmd window in admin mode):
quickbuild.bat -j12 --hash optimization=space address-model=64 pwiz_tools\BiblioSpec

quickbuild stops very fast with the following error:

[image]

I tried with both, Visual Studio 2013 as suggested on the Skyline homepage and  Visual Studio 2017 as indicated here.

Do you have any idea what I am missing?

Best,
Juergen

chambm, 2020-09-03:
Definitely needs VS2017, where did you see it say 2013? Try running from the `Developer Command Prompt for VS 2017` which should be in your start menu. Do you have VS2019 installed as well?

JB91451, 2020-09-03:
I saw the error with both versions (exactly the same error). Yes, VS2019 is also installed and required for some other software on the same computer.

JB91451, 2020-09-03:
The information about 2013 came from:
https://skyline.ms/wiki/home/software/BiblioSpec/page.view?name=Download%20and%20build

chambm, 2020-09-03:
OK fixed that page. Bjam will default to the highest VS installed although that shouldn't actually block building of bjam, just the .NET projects. I build with VS2019 on my personal machine and it runs ok. Try adding `toolset=msvc-14.1` to the command-line.

chambm, 2020-09-03:
Also you might try running from VS2019 developer prompt instead of VS2017. Bjam can be built with any compiler, that's really a bootstrapping step.

JB91451, 2020-09-03:
I tried to run the command with the additional toolset argument (is there a "-" / "--" required in front?) and both VS versions from the developer command promts now but the error was exactly the same. I guess that the position of the toolset argument is not important?

chambm, 2020-09-03:
No `/` required. Try `cd`ing to libraries\boost-build\src\engine and running `build.bat vc141`

Then go back to the pwiz root and rerun quickbuild.

JB91451, 2020-09-03:
Unfortunately, already this step produces an error:

[image]

JB91451, 2020-09-03:
Could this be a corrupted download?

chambm, 2020-09-03:
I can't imagine you're in a working VS2017 command prompt if you are getting `'cl' is not recognized as an internal or external command`. Does `cl` work by itself on the cmd prompt?

JB91451, 2020-09-03:
Do you mean just running cl from the .\engine folder? This does not work.

I tried to do the same with VS2019 again and there some kind of build is happening. However, the generated build-nt-x86 folder contains an empty BiblioSpec subfolder (except a file named "version").

chambm, 2020-09-03:
Running just `cl` from any directory should give you the compiler's usage information (and tell you that it's in the PATH). Usually Boost.Build can find the VS binaries by itself without needing to be in your PATH already, but having it in your PATH is another way to do because it inherits the current environment.

Try `build.bat vc142`

JB91451, 2020-09-03:
Ok, in between I installed git because when I tried the clean.bat there was an issue that git would not be a valid command.

I tried "build.bat vc142" with both VS versions: 2017 still does not recognize while in 2019 it works.
Should I try the quickbuild.bat from the root dir with that version again (and save a log file)?

chambm, 2020-09-03:
Yes, with the toolset=msvc-14.1 argument.

JB91451, 2020-09-03:
It run through but as above, the folder is empty. Please find the log below.

[quickbuild_log.txt](https://github.com/ProteoWizard/pwiz/files/5170243/quickbuild_log.txt)

chambm, 2020-09-03:
There were errors. Get rid of --hash so I can see that it's actually running with msvc-14.1. . Are you running from the VS2019 prompt? If so try from the 2017 prompt and from the regular command prompt. It might be using the wrong SDK.

JB91451, 2020-09-03:
Did so. But both failed with the "'cl' is not recognized as an internal or external command, operable program or batch file." error

[quickbuild_log_VS2017.txt](https://github.com/ProteoWizard/pwiz/files/5170281/quickbuild_log_VS2017.txt)
[quickbuild_log_WinCmd.txt](https://github.com/ProteoWizard/pwiz/files/5170285/quickbuild_log_WinCmd.txt)

JB91451, 2020-09-03:
... and from VS2019

[quickbuild_log_2.txt](https://github.com/ProteoWizard/pwiz/files/5170336/quickbuild_log_2.txt)

chambm, 2020-09-03:
What's the output from `set` in your VS2017 prompt? Can you run `cl` and get the compiler usage?

JB91451, 2020-09-03:
The output from set is given below. What do you mean with run cl and get the compiler usage? When I run it in the VS2019 prompt it returns:
C/C++ Optimizing Compiler Version 19.26.28806 for x86

[log_set_VS2019.txt](https://github.com/ProteoWizard/pwiz/files/5170490/log_set_VS2019.txt)
[log_set_VS2017.txt](https://github.com/ProteoWizard/pwiz/files/5170491/log_set_VS2017.txt)

chambm, 2020-09-03:
It should give you a bunch of help text (and the version as you quoted). What do you get from running it in the VS2017 prompt?

JB91451, 2020-09-03:
still the "is not recognized" error

JB91451, 2020-09-03:
The help text is only a one-liner with the usage: information in 2019

chambm, 2020-09-03:
Then I'd say you have a corrupt VS2017 installation. Sorry, just `cl` gives the version and one-liner usage. You have to add `/?` to get full help. Reinstall VS 2017 and make sure the VS 2017 command prompt can run `cl` :)

JB91451, 2020-09-03:
Ok, I will try that tomorrow morning and keep you updated about the result. Thank you very much for the extensive help that you provided today!

Best regards,
Juergen

JB91451, 2020-09-04:
Dear Matt,

Thank you for your help again. Indeed a broken VS2017 installation was part of the solution. I reinstalled it today and the solution you mentioned earlier worked than. I will shortly summarize it in case someone else struggles from the same issue:

On my system, where VS 2019 and VS 2017 are installed in parallel, the build process failed at an early time point with an error indicating "Error building bjam". 
1)
In order to build the boost library separately, one has to change the directory to: 
.\libraries\boost-build\src\engine\
and run:
build.bat vc141
2)
afterwards change directory back to the root and run the quickbuild.bat with an additional toolset argument:
quickbuild.bat -j1 optimization=space address-model=64 pwiz_tools\BiblioSpec toolset=msvc-14.1
3)
The BiblioSpec executables should be located in a subfolder like this one .\build-nt-x86\msvc-release-x86_64

I also have a last question: Can the executables be moved to a different location and can I delete stuff outside the build-nt-x86 folder in order to release some drive space?

chambm, 2020-09-04:
If you don't want to build again, then sure, you don't need the source code. Does your question mean you only wanted the executables to begin with?

JB91451, 2020-09-04:
Yes, we need them in order to build/convert a blib spectral library from mascot searches to the ms2 format and than import this to spectrast.

chambm, 2020-09-04:
Well, I guess it's a learning experience. :) On this page, `https://skyline.ms/wiki/home/software/BiblioSpec/page.view?name=Download%20and%20build`, I think you wanted this section:
```
Pre-compiled binaries
64 bit Windows binaries (BlibBuild.exe, BlibFilter.exe and BlibToMs2.exe) are available from the ProteoWizard automated build and test system. (If you encounter a "Log in to TeamCity" page, just click the "log in as guest" link to proceed.) Note that this download does not provide a proper installer: it's intended for updating existing installations only. It likely will not function if unzipped to a bare directory.
```

chambm, 2020-09-04:
Oh, except you wanted BlibToMs2. Which won't be in that zip file. Hmm. I'll see about adding that to a separate artifact.

closed 2020-09-04 by chambm

JB91451, 2020-09-04:
Oh, I even saw that page but was confused by the statement:
"...it's intended for updating existing installations only. "

As you said - it's a learning experience than. 

Thank you once more!

chambm, 2020-09-04:
Oh actually it does have BlibToMs2, I must have addressed that some time ago. It does run by itself but is missing most of the vendor reader DLLs and the side-by-side VC runtimes. You could put it in a regular ProteoWizard installation to enable reading RAW/WIFF/etc. You have the VC runtimes installed by having VS installed though.