ProteoWizard/pwiz #1216: Different precursor assignment between msconvert 3.019 and msconvert 3.20

prvst, 2020-09-08:
Dear @chambm 

We tried to reproduce some results from previous analyses, which include TMT quantification, and we noticed major differences in the output. The major issue was the lack of any TMT quantification for a considerable number of PSMs. After some investigation, we noticed that the divergence happens when we use different msconvert versions (3.0.19127 and 3.0.20236).

I converted the same Thermo raw file using version 3.0.19127 and version 3.0.20236. When using version 3.0.19127, we see that msconvert assigns the scan 2785 (MS 2) to the precursor scan 2776
[image]

When we look at the same spectrum, from the file converted by version 3.0.20236, we can see a different assignment. This time the precursor scan is the scan 2775.
[image]

Both 2775 and 2776 are MS1 scans, and 2776 is the closest to this MS2 scan. The reason why we see no TMT quantification for the PSM matching to this spectra is that the isobaric labels are in the scan 2776, not 2775.

Should we stop using the `SpectrumRef ` attribute for linking spectra?
I would appreciate if you could advise us on the best way to link scans using the latest msconvert version (MS3 -> Ms2 -> MS1).

Thanks

chambm, 2020-09-09:
I think you must have a typo because you say that the isobaric labels are in 2776, which you say is an MS1 scan. The isobaric labels should only be in MS2+ scans. I recently changed the spectrumRef calculation for Thermo to use the Master Scan scan header attribute (if available) rather than the heuristic we have been using. For SPS TMT MS3 workflows it works much better. And even more recently (https://github.com/ProteoWizard/pwiz/pull/1214) I fixed another bug related to this.

prvst, 2020-09-09:
Sorry, you are right. I guess that we just want to be sure that the different precursor assignment is not the product of a bug or something like that.

chambm, 2020-09-09:
But if you are seeing incorrect assignment of MS3s to MS1s then I want to know about it (except the MS1 precursor will be given as a separate `<precursor>` element at the end of the list, so that you can trace the entire MS3 precursor hierarchy even if the MS2s aren't stored in the file, such as in MS3 inclusion list methods).

prvst, 2020-09-09:
We are still trying to figure out what the issue is, but we think that there might be an issue with the MS3-> MS1 linkage. IN that particular case above, the precursor scan seems to be wrong. If you want, we can move this discussion to the email. I think that Alexey might want to add to the discussion as well.

chambm, 2020-09-09:
If you can send me the RAW file I'll look at what's going on.

prvst, 2020-09-09:
Could you give me an email to share the file? I'll upload it to Box

chambm, 2020-09-09:
matt.chambers42 `@` gmail.com

chambm, 2020-09-10:
Thermo isolation window not reported the same between precursors at top level vs. second level (e.g. an MS3's MS1 isolation width is not the same as its MS2's MS1 isolation width).

Fix SPS masses not being parsed when they are in separate header items like "SPS Mass 1:", "SPS Mass 2:", up to 20.

mobiusklein, 2021-01-06:
I think this also affects product ion triggered MS/MS spectra, leading to an MS2 spectrum having its `spectrum/precursorList/precursor@spectrumRef` set to another MS2 spectrum, instead of the preceding MS1?

First MS2
```xml
        <spectrum index="29" id="controllerType=0 controllerNumber=1 scan=10079" defaultArrayLength="3358">
          <cvParam cvRef="MS" accession="MS:1000580" name="MSn spectrum" value=""/>
          <cvParam cvRef="MS" accession="MS:1000511" name="ms level" value="2"/>
          <cvParam cvRef="MS" accession="MS:1000130" name="positive scan" value=""/>
          <cvParam cvRef="MS" accession="MS:1000128" name="profile spectrum" value=""/>
          <!-- snip -->
          <cvParam cvRef="MS" accession="MS:1000796" name="spectrum title" value="LUM2_06054_MNM.10079.10079.2 File:&quot;LUM2_06054_MNM.raw&quot;, NativeID:&quot;controllerType=0 controllerNumber=1 scan=10079&quot;"/>
          <!-- snip -->
          <precursorList count="1">
            <precursor spectrumRef="controllerType=0 controllerNumber=1 scan=10049">
              <isolationWindow>
                <cvParam cvRef="MS" accession="MS:1000827" name="isolation window target m/z" value="1249.00927734375" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                <cvParam cvRef="MS" accession="MS:1000828" name="isolation window lower offset" value="1.0" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                <cvParam cvRef="MS" accession="MS:1000829" name="isolation window upper offset" value="1.0" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                <userParam name="ms level" value="1"/>
              </isolationWindow>
              <selectedIonList count="1">
                <selectedIon>
                  <cvParam cvRef="MS" accession="MS:1000744" name="selected ion m/z" value="1248.007202148438" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                  <cvParam cvRef="MS" accession="MS:1000041" name="charge state" value="2"/>
                  <cvParam cvRef="MS" accession="MS:1000042" name="peak intensity" value="3.887858498535e06" unitCvRef="MS" unitAccession="MS:1000131" unitName="number of detector counts"/>
                </selectedIon>
              </selectedIonList>
              <activation>
                <cvParam cvRef="MS" accession="MS:1000422" name="beam-type collision-induced dissociation" value=""/>
                <cvParam cvRef="MS" accession="MS:1000045" name="collision energy" value="35.0" unitCvRef="UO" unitAccession="UO:0000266" unitName="electronvolt"/>
              </activation>
            </precursor>
          </precursorList>
```
with Master Scan Number "10049"

Second MS2
```xml
        <spectrum index="30" id="controllerType=0 controllerNumber=1 scan=10080" defaultArrayLength="8814">
          <cvParam cvRef="MS" accession="MS:1000580" name="MSn spectrum" value=""/>
          <cvParam cvRef="MS" accession="MS:1000511" name="ms level" value="2"/>
          <cvParam cvRef="MS" accession="MS:1000130" name="positive scan" value=""/>
          <cvParam cvRef="MS" accession="MS:1000128" name="profile spectrum" value=""/>
          <!-- snip -->
          <cvParam cvRef="MS" accession="MS:1000796" name="spectrum title" value="LUM2_06054_MNM.10080.10080.2 File:&quot;LUM2_06054_MNM.raw&quot;, NativeID:&quot;controllerType=0 controllerNumber=1 scan=10080&quot;"/>
          <!-- snip -->
          <precursorList count="1">
            <!-- Note this refers to the previous MS2 -->
            <precursor spectrumRef="controllerType=0 controllerNumber=1 scan=10079"> 
              <isolationWindow>
                <cvParam cvRef="MS" accession="MS:1000827" name="isolation window target m/z" value="1249.00927734375" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                <cvParam cvRef="MS" accession="MS:1000828" name="isolation window lower offset" value="1.5" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                <cvParam cvRef="MS" accession="MS:1000829" name="isolation window upper offset" value="1.5" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                <!-- Unclear how the original value this draws from in SpectrumList_Thermo.cpp is set  -->
                <userParam name="ms level" value="1"/>
              </isolationWindow>
              <selectedIonList count="1">
                <selectedIon>
                  <cvParam cvRef="MS" accession="MS:1000744" name="selected ion m/z" value="1248.007202148438" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                  <cvParam cvRef="MS" accession="MS:1000041" name="charge state" value="2"/>
                </selectedIon>
              </selectedIonList>
              <activation>
                <cvParam cvRef="MS" accession="MS:1000422" name="beam-type collision-induced dissociation" value=""/>
                <cvParam cvRef="MS" accession="MS:1000045" name="collision energy" value="30.0" unitCvRef="UO" unitAccession="UO:0000266" unitName="electronvolt"/>
              </activation>
            </precursor>
```
with Master Scan Number "10079"

This was converted to mzML using ProteoWizard release 3.0.20299.

chambm, 2021-01-06:
Huh. I've never thought of an MSn scan triggering another MSn scan of the same n. So, I need to fall back to the old heuristic if master scan is the same ms level?

mobiusklein, 2021-01-07:
It might be simpler to say if the ms level of the precursor matches the product, try the precursor's precursor until the ms level drops? I don't know which is more accurate vs. computationally expensive. 

This is complicated when an instrument seems to interleave acquisition of MS2s from one MS1 scan with the acquisition of the next MS1. This makes some amount of sense for instruments with two mass analyzers (e.g. TIMS-TOF, Thermo Tribrids with both Orbitrap and Ion Trap).

edwardsnj, 2021-04-30:
I'm just now seeing this issue due to updating my msconvert version, for a spectral file conditionally triggered MS2 spectra only (EThcD triggers HCD), but in a section of the file where no triggering is taking place. The series of XML tags tells the story...

        <spectrum index="21" id="controllerType=0 controllerNumber=1 scan=22" defaultArrayLength="559">
        <spectrum index="22" id="controllerType=0 controllerNumber=1 scan=23" defaultArrayLength="2816">
        <spectrum index="23" id="controllerType=0 controllerNumber=1 scan=24" defaultArrayLength="37">
            <precursor spectrumRef="controllerType=0 controllerNumber=1 scan=22">
        <spectrum index="24" id="controllerType=0 controllerNumber=1 scan=25" defaultArrayLength="506">
        <spectrum index="25" id="controllerType=0 controllerNumber=1 scan=26" defaultArrayLength="124">
            <precursor spectrumRef="controllerType=0 controllerNumber=1 scan=23">
        <spectrum index="26" id="controllerType=0 controllerNumber=1 scan=27" defaultArrayLength="668">
            <precursor spectrumRef="controllerType=0 controllerNumber=1 scan=23">

scan=24 is MS2 should have precursor scan scan=23 not 22, scan=26,27 are MS2 should have precursor scan=25, not 23. 

Previous version(s) got this correct (pwiz version 3.0.18320 gets this right, 3.0.20287 is the above). No none of these three MS2 spectra, did the second (triggered) spectrum happen.

Any insight? I'd be happy to provide a .raw file...

edwardsnj, 2021-04-30:
@prvst did you figure out a workaround? I can ignore the precursor spectrumRef and use my own simple-minded heuristic, but I hate having to put temporary exceptions into my code-base (I always forget to remove them later...) I'll test a more typical proteomics RAW file today to see if it also has the issue. Based on my example above, I'm worried this occurs even without any triggering logic. 
@chambm any progress on this? its been in the wild for six-months...
Thanks...

prvst, 2021-04-30:
@edwardsnj At the time, we've made it work with 3.0.20253, using the precursor spectrumRef information from the MS2. I haven't checked again since September or October, but I would be worried if you see mismatches again with the latest version. We have been telling people to always use the latest msconvert version because, for reasons I'm unable to determine yet, the isobaric label quantification will fail completely with some previous versions.

chambm, 2021-04-30:
Try this build to see if it's fixed @prvst @edwardsnj : https://teamcity.labkey.org/repository/download/bt83/1396982:id/pwiz-setup-3.0.21120.048532aa7-x86_64.msi?guest=1

chambm, 2021-05-03:
@edwardsnj As far as I can tell it's working as intended now on your file. As long as Master Scan points at a MSn-1 scan, that will be chosen as the precursor. It won't necessarily be the most recent MSn-1 scan. For your example, MS2 scan 24 has "Master Scan: 22" and scan 22 is an MS1, so it is chosen as precursor. Likewise for scans 26 and 27.

Using the most recent MSn-1 scan was the behavior before we started using Master Scan. There are arguments to be made for both approaches, but I think the Master Scan wins out because Thermo's metadata basically points to it as the precursor scan.  If you definitely need the most recent MS1, it's probably best if you make your heuristic permanent. However, like airplane exits, the closest MS1 may actually be AFTER the MS2, right? :)

edwardsnj, 2021-05-03:
Hmmm. I'd rather use the correct value - the instrument knows where it got the precursor from and pwiz has access to that metadata. I'm wondering why the scan numbering is as it is, then, but again, pwiz is just representing what the raw file says. Time/causality can't go backwards, but the scans are not instantaneous, and MS1 scans might be faster or slower than MS2 scans. Ugh. Thanks for looking into this.

edwardsnj, 2021-05-07:
OK, I found a more direct example of the non-trivial behavior with MS1 -> MS2 -> triggers -> MS2 in the file I sent you earlier. My understanding of how the instrument rule is set up: MS1 -> EThcD MS2 -> trigger HCD MS2 in the same cycle on same precursor as EThcD spectrum if ion XXX is observed. I don't think the change you made is correct here, but please tell me if you think otherwise. 

Scan 59: MS1
...
Scan 68: MS1
Scan 75: MS2, EThcD, master scan: 59
Scan 76: MS2, EThcD, master scan: 59
Scan 77: MS2, HCD triggered from Scan 75. 

Old versions use most recent MS1: so scan 75, 76, 77, all say precursor scan is 68. 
3.020 says 75 has precursor scan 59 and 77 has precursor scan 75.
modified version says 75 has precursor scan 59 and 77 has precursor scan 68. 

At this point, I *think* 3.020 is the "most" correct for scan 77 and the precursor element indicates the master scan value. To get the precursor scan I think it'd be necessary to "follow" the precursor scan link of scan 75, because it is the same level as 77. For this type of analysis a master scan is not the same as a precursor scan. 

Sorry this is so confusing, I guess it is possible that even the concept of a precursor scan is moot (or not recorded in the same sense in the RAW file). 

Please let me know if you can't follow the example.

Thanks...