ProteoWizard/pwiz #1236: Cannot convert from wiff to mzML using MSConvertGUI

tomgrove20, 2020-09-22:
I have a wiff file (MRM) from a Sciex 6500 triple quad. I would like to plot this in R, so I am trying to convert to mzML using pwiz MSConvert (GUI in Windows). I have both the wiff and wiff.scan files (same name) in the same folder. 

The GUI looks like this:
[image]

And the progress bar looks like this:
[image]

That screen has not changes in 25 minutes. 

I attach the wiff and wiff.scan below. I'd be really grateful for any advice on how to make this work.

[test-wiffs.zip](https://github.com/ProteoWizard/pwiz/files/5263668/test-wiffs.zip)

Thanks and sorry for the trouble!

chambm, 2020-09-22:
This WIFF has ~32 samples in it so msconvert would normally create 32 output files, one for each sample. The samples don't have unique names. I'm not sure why it's hanging on one of the samples, possibly one of them is corrupt. The GUI doesn't have a way to select which sample(s) you want to convert, but the command-line msconvert.exe does. Use the --runIndexSet argument for that, like `msconvert test.wiff --srmAsSpectra --runIndexSet 1-32` (index is 0-based). I tried it myself and the command-line exe doesn't get stuck like the GUI.

tomgrove20, 2020-09-22:
@chambm command line worked perfectly, thanks so much!

closed 2020-09-23 by chambm