ProteoWizard/pwiz #1256: UNIFI conversion to mzML using DDA data

MHJena, 2020-10-05:
Hey,

currently it seems like exporting mzML files from a UNIFI database is not supported for DDA data ("only MSe and HD-MSe data supported at this time).

Is there any other way to export a mzML containing MS1 AND MS2 information from UNIFI? Is there a systematic reason why DDA data is not supported?

Thank you very much!

chambm, 2020-10-05:
Yes there's a reason: the UNIFI API doesn't support spectra access for anything except MSe.

gzu300, 2022-05-03:
The unifi reference api shows under Sample results there are 2 endpoints:
'/sampleresults({resultID})/spectra/mass.dda'

Can these 2 endpoints be used to download dda data?
[image]