ProteoWizard/pwiz #1263: Thermo RAW to mzML: reference mass signal missing

breidan, 2020-10-07:
When converting MS1 data from our Thermo Orbitrap Elite to mzML with `msconvert` then the signals of lock masses do not seem to be converted along. I assume that to be true for all Orbitraps. In Thermo's QualBrowser the lock masses are normally not displayed either but one can tell QualBrowser to display these "reference and exception masses". Is the same possible for `msconvert`?

In general, being able to have the lock masses signals in the mzML would enable one to use them to analyse measurement system stability. In particular, I have the case that I have a compound which is only 0.07 amu away from the lock mass and when that compound is eluted from the LC the OE seems to select the compound m/z as reference mass. The result is that the respective compound's peak is not present in the mzML file.

chambm, 2020-10-07:
That's an interesting issue. Can you send me the affected RAW file and the scan and m/z to look at?

breidan, 2020-10-08:
@chambm,
the attached raw file exemplifies that behaviour. Acquired on an Orbitrap Elite, APCI, FTMS1, lock mass m/z 149.0233 (phthalic anhydride ion). The compound of interest has m/z 149.0961 and starts eluting around scan# 1610. At scan#1611 the reference mass is 149.0233 and at scan#1612 that changes to also include 149.0961. Displaying that file in QualBrowser without including reference masses will hide the peak which elutes there. 
I converted the file with `msconvert` (I have to admit it is an older version and since I am working from home I cannot access the version number) to mzML,  and loaded it into R using `MSnbase`. A chromatogram of m/z149.0961+-10 ppm comes out mostly empty. Same for m/z149.0233+-10 ppm. 
Now, I assume it is the conversion from .raw to .mzML where it goes wrong. If you could have a look that would be much appreciated.
[001_QC.zip](https://github.com/ProteoWizard/pwiz/files/5345758/001_QC.zip)

chambm, 2020-10-09:
Try the latest pwiz 20282 or later (Windows build, not Docker).

breidan, 2020-10-09:
@chambm,
great. I will test that on Mon and let you know if it worked.
Thanks for the help.

breidan, 2020-10-12:
@chambm,
I converted 001_QC.raw with `msconvert 3.0.20283` for Windows and the behaviour is still the same as with the old `msconvert 3.0.6585`. Scans 1610&1611 contain m/z 149.0961+-5ppm, scan 1612 does not. The data was acquired as centroids. Nevertheless, I also ran `msconvert` with the `peakpicking` filter. Same outcome for both.
Did I miss something?

breidan, 2020-10-12:
I found this in the documentation of Thermo's RawFileReader:

> The property “bool IncludeReferenceAndExceptionData { get; set; }” will change the operation of many calls which return MS data.

Do not know if this helps.

chambm, 2020-10-12:
Try the latest version now. Sorry about that.

breidan, 2020-10-13:
@chambm ,
it is working now. I see the lock mass and I do see the m/z 149.0961 in every scan it should be.
Thanks for your help. And ProteoWizard really rocks.

breidan, 2020-10-15:
@chambm,
any chance that this change will make it into docker any time soon?

chambm, 2020-10-16:
I published it yesterday.

closed 2020-10-16 by chambm

breidan, 2020-10-16:
@chambm,

I hate to say it but I just pulled the latest docker image and it did not include the reference masses during conversion.

chambm, 2020-10-16:
I just tried it and I do get the reference peaks with the docker image. Are you using the image with digest 7d1345944ad1?

breidan, 2020-10-17:
@chambm,

sorry, my bad. I pulled the latest image assuming that the older image is completely overwritten. Apparently that was not the case. After removing all images and pulling the latest, it now works as expected.
Thanks again for your help.