ProteoWizard/pwiz #1269: Extracting the precursor m/z and charge state from a Thermo RAW file

MarcIsak, 2020-10-09:
Hi, 

I wonder if there is a way to use ProteoWizard to extract the precursor m/z and charge state for different MS1 scans of Thermo Raw files? 

When I look at a Raw file in XCalibur, it is possible for me the display the precursor m/z and charge state in an MS1 scan (see image). It would be great if this info could be extracted  into a text file or similar.  If not with ProteoWizard, is there another similar software/algorithm that has this capability? 

[image]

Best,

Marc

chambm, 2020-10-26:
There's no way to do this with ProteoWizard. It could be added fairly easily, but would need additional user-interface work, e.g. some kind of "extra" level of verbosity that the user could ask for, because otherwise it would add a lot of size to the file without benefit for most users. Not something we have planned at the moment, but PRs are certainly welcome.

btemoshir, 2024-12-11:
I would also like the functionality in MSconvert to be able to write the charge states for each peak in the mzML file. This data is saved by default in the Thermo raw files, but I don't know where it goes after conversion to mzML.

grandrea, 2024-12-30:
hello, I am looking for a similar feature. I think there is a lot of demand from machine learning parts...

grandrea, 2024-12-30:
+1 for this functionality, it can enable a lot of machine learning applications

OnTheTomSci, 2025-07-12:
+1 again for add this capability needed for machine learning approaches