ProteoWizard/pwiz #1273: boost error

ococrook, 2020-10-13:
I got the following error on windows 10, with the version downloaded from proteowizard currently on the website. Let me know if you would like more information. 

$ msconvert 200705_PhosB_2pHDMSE_GF_0sec_01_vial01.raw
format: mzML
    m/z: Compression-None, 64-bit
    intensity: Compression-None, 32-bit
    rt: Compression-None, 64-bit
ByteOrder_LittleEndian
 indexed="true"
outputPath: .
extension: .mzML
contactFilename:

filters:

filenames:
  200705_PhosB_2pHDMSE_GF_0sec_01_vial01.raw

boost::filesystem::create_directories: Invalid argument
Please report this error to support@proteowizard.org.
Attach the command output and this version information in your report:

ProteoWizard release: 3.0.10827 (2017-5-11)
ProteoWizard MSData: 3.0.10827 (2017-5-11)
ProteoWizard Analysis: 3.0.10827 (2017-5-11)
Build date: Nov  1 2017 15:08:32

chambm, 2020-10-13:
You downloaded the most recent version, but you're not executing it with your command above. Try the absolute path to msconvert.exe, like `"c:\Program Files\Proteowizard\ProteoWizard 1.2.3\msconvert.exe"`

ococrook, 2020-10-13:
ahh yes, I see there is a file path issue - thank you!

closed 2020-10-13 by ococrook