ProteoWizard/pwiz #1294: Format conversion from timsTOF DDA file to mzXML

Wenhhao, 2020-10-22:
Hi Matt,

I converted a timsTOF Pro DDA file (TDF) to mzXML using MSConvert GUI (Version: 3.0.20287-769529fa4) with "Bruker TDF Defaults" filter parameters. But I just find out "ion mobility upper/lower limit" instead of "precursor ion mobility", which could not be recognized by the search engine I used. Is this caused by the peak width of precursor ion mobility? In other words, whether the top and bottom limitations of precursor ion mobility are reported rather than the peak value?

One of the scans' information is shown as following:

><scan num="7411"
>      scanType="Full"
>      centroided="1"
>      msLevel="2"
>      peaksCount="242"
>      polarity="+"
>      retentionTime="PT1184S"
>      collisionEnergy="30.620338983051"
>      basePeakIntensity="179.0"
>      totIonCurrent="1721.0"
>      msInstrumentID="1"\>
>  <precursorMz precursorIntensity="0" precursorCharge="2" activationMethod="CID" windowWideness="2.0"\>609.25938574874294</precursorMz\>
>  <peaks compressionType="zlib"
>         compressedLen="2350"
>         precision="64"
>         byteOrder="network"
>         contentType="m/z-int"\>eJxFl3k81FsbwL3csiQibyFbJFNXZamrhTyyFSn7XtZslcxYw1iGMYzhRaR6W4gpRAh1ozTZKglXRGlxKVQ3Ialwue/n7Zxj/vt+nv05z3l+Z8CNmyQww7UCZb7//8BtoVFG/+8Bwp41Nn8Uvegh7P2A923sqihh3yymicEaKcL+Yd2VzsfrYR3iY0YtPPqIPtgD4oJHyjJ2OeAZh3hS0vFhzjPYi3kqN6TNNJdwYOLe5ohOBuggf4FMi/IoGQtwRv6oEvMlp9Jb4QgPseQGpaWTg+CK9Kn5X/1fn7AEJyyvT7bZyPUn+VLvXwx9GZ0Bx1A8aoN5WeBTHfAaQPxiU5ue/3LYgfRpK/ke+Ys+BUOkT7N9wfK+7kb80f6YmVBLGwP2CsRdaqx3lFzwy0M8qBG6+uM7cMf6Q79kj5o6ghni4J1q2la9M+CJ8g32aH3WRvGBoPGfHHLf4EqXtDTYovihKXK7zu+nk/hhjvK3KFZypL9hngGK/m9HwB4zzX1+WZ4FHEL+T6osiVBQMAB3pZ8cyd/UtyTlIHii+qOUdPcHm3oQ/1El9060V9DA52cBEE1rWcIVNyT10EWoA7KXTMEVnQ9d8sjprDEjcMPyrPYImQPBcBL1g35KynJN0TBYxGHer9WsWALeKD79Mk+1cLAcgjIwf5eS+FYKbih/esOL4w9amGCD4/VLSqWubgFbJI+R/dEkeCcRPDAXWOQUeEXDYRQvpk465EnVPTJfMV8WUoRMC0Af85TXqRQXIPMWM5v7ZqvQZjiO8ouZu5rd0bcdtiL9WIGCnPaTmuCF/McKO2jLRxfDb1i+rOJQiP13Yh+7XnRVU89y0t9Y7eKmw4bbwQHFi7XJtfc0LQEPwg0nz4uWgzXWt5md1dThA44QYs+adZ33hsEd+/f8e80ZxVXggu29+TSM1IPBH51fnHzNsCplN1gC5tEp9rQ2ySduR7VWpcwOsMe8r1f3/fQYma84Ou+/dUbdkIHl9Gd3fTyXQpgG5tn8hfS8RX8x1FJv53gwxRzfYDy1cBJS+zD3BViqz0EyBfPXieERGngqIWbI9tZ2nF30x3yqLc3937zielLaQz7nvwF3nF/GFeF5UyXYhvULnOfWlOqR+xzHbZDWtnIDb6xfIqu+ueY0mde4Snc7g7lbYIs4XuamoVPGF/BA+vHbimtEDSvJvonX8xAcczgKNpj3Fc4srGOCO+pvvFXVp4a6XjiK2elKjGu4KlhjPlTlLawWDraYPS6HFspTyPnHH8lXGNYsJPXH08fudruvhRN5iDMr7ntl94EJludflBAXVwcLzJejzBrNNcCB95MZImJyZY9qgYX6zRATaps02AFUPsyKXxX/+g0OoXoZkhuTfKujwAvbO98ucjCgkX3IiCozsyidJ/eBQR+JFlzogQzsP0ZB/Sk3leTPiH+jtyI5E3zyME+XqbM04Cj2z5yPaXMwIPuQcWFUd2Jo06J9g4DmwVxX8EPnz2jym27cPUbmhdHCrzOV6QiRqH+MRz8eJ1RvJf1kPHn/sdWkinCC8FDcmW4TcOVhnjXRUvaGs9GIRVYuf+IqDlnJmDf2BiRnkPNNENlzSSViCckvYbmo0pJifTJ/CWL/+vyqjg9OhWBee/X5awdIpWA28JSSmFm0F7efWOssDkaEr2uU5M3CdsySwkIisQIQrYFZqev33n44hvqRsJI+ulVzJwDWX8VXoqPIAzec7ypW7o0+bfDC8tVcG4GlaXAQs7LVYdWX5WQfJqzLSajV6YK9mFtdeovM/gN26PwTNYcK5RTcyP1J1GsNuBuRQuYvcfctavOKMTiC8kuEfpXZ5VJAwXIzR5cD2kD2daJf7BNbHj/pR2Lr2U/jvChwxPFaH+drv0wFbSRnSjQG8JcfJPUwJb7Sli2bJPGZMgORrjOrwBzZM80uvi6w2gJJhB8+XC1QDNEoPtPcT7zI5E/yfWBa/c2K4xcDF8zWZdfvOAmDL+EO3bwgJQgcQOyUd0FCTBZ2Y7lzyzFF/g/giuOxhaS6ejLBCZ0H8/nclpHAFWCH4385/SbGRYXUnySXtLN6G4u8p5ICJKLNw8chvBPzr6ZB3/8EKrJPoinbbCg/TPZVUui3y9ReVTDH/iJVfNt9T4OXEmbjftmoDtDF8viaV8zHbqS/LIG1Rv1iz8n3hLW33stiLgEMkH+WWV5qostR4Ixiru2rDwqDtO2IzYUbKR7fYT/Wty5LuXlXnew/lvWD59K3soExjnmOLvL+BhzA+s4ujuuduOCA83E+87hjqAkceZirFItYXHIfWZ6b0vQX6sES68dF6bf3LCPnwSrJK7ldZUz2O+sbN/I5zQusECc7wkIz/SPpf/LgGNU6N5TcpxR73uYPCqVggPnc47nzLankPqTcKN7U5KMLjiiflK6xGNuBB8Q/m3/Cv6WgFPYgffYvMiPVxnokHjugfkt4cid5D7MjT3iEyBiS9xy7pDtYoPw8+APm8XOq7dWwD8uvid7xDSyHMMKUgb+0LCGqAvOBAIXTAHY43o23g42Hg8Ea9ZNdJa9zRPQq2b/sOmF7fX8qHMf+6lQTQjudwArLB39Eb9kwA8bY34dp+SGBSPK+Y09+iP81diP5/qc6Xzs2UX6JvJdSsz57f1J2J/OQenbXKwU/Duhh/XPnF+41pZH3Ueq5Nx1++8LBBdsPltm8mnpI+sPRFlv/7NQucr84Ls1XVWT4SH85tFp3uaVsOIQ5etcGYVob2W+ckpbElIBcsMH+rvENqBV2QnYiZgUxTeAHZh9mo2k7yhXwGUB8Xaor3HgDuR+cG3TGbY1t5L3KqRKPTsnrBaoGZvXrpSWPyLxyar8b9IyEwgnMdf+OMFUtJPuBU6dVqaaVTb5/nLrj8g1N+Yv1NUr27hkSJN8LzoPul++/uAPgfD4qqUgKmIIN9jfLr2W1uZHsi7QPUeHl65bCHmSfNsnzuVkhTfqZLhhX8d6oHJyRfTqtuVQw4iD4If30yEaZj+Le5D5k7HhHyX4hTf4fZOyc1LtQufh/LCP/9/U2t/vJ+z2TL529mssj8swfozGy8JRwttOsYZb7CsI5SrkjQSPjhHN7es32arsQPhtWfGl2+O0it1eqtk+8JHwuaGZ+QH0b5n8ArEVUvA==</peaks\>
>  <nameValue name="ion mobility lower limit" value="0.86384915849942578"/\>
>  <nameValue name="ion mobility upper limit" value="0.89118545105577796"/\>
></scan\>

Best regards,
Wenhao

Wenhhao, 2020-10-22:
Sorry but I didn't work in this field two years ago, umm... so I am not sure whether the latest mzXML schema support ion mobility, at least I can't find that one on google.

By the way, I do find out ion mobility in pwiz mgf file, so I think pwiz mgf support ion mobility, is it right?

chambm, 2020-10-22:
Yes, the TDF default profile will put ion mobility in the MGF TITLE string. It's up to the reading software to interpret it because the TITLE string is totally free text.

Wenhhao, 2020-10-22:
I see, thanks a lot for your reply.

closed 2020-10-26 by chambm