ProteoWizard/pwiz #1301: What encoding does msconvert use when writing mzML m/z and intensity arrays?

wkumler, 2020-10-29:
Hi, thanks for all the awesome tools that you've built!

I'm currently trying to parse the raw XML format that mzML files are written to by `msconvert`. Much of this is entirely intuitive, and the [mzML format specs](http://www.psidev.info/mzML) have been super useful decoding most of it. However, I'm struggling to understand how the _m/z_ and intensity data arrays are encoded within the document.

I initially presumed that this complication was due to the use of zlib's `deflate` algorithm to compress the files via the `msconvert -z` option. However, even when processing the files without compression, I'm finding the data arrays difficult to parse. It *seems* like they're encoded via base64, but none of the decoders I've tried have been able to parse it back into a numeric representation.

This is perhaps a novice question, but I'm trying to understand what happens to my data along every step of the mass-spectrometry pipeline. I'm happy to provide some subset of my own data to look at, but the same encoding can be seen in the example mzML provided in the spec sheet: http://proteowizard.sourceforge.net/example_data/small.pwiz.1.1.mzML.

chambm, 2020-11-02:
The mzML arrays are either 64 or 32-bit floating point or integer arrays. The datatype is in a binaryDataArray cvParam. It's almost always 64 or 32-bit floating point. To represent this binary data in XML it uses base64, and to save space the binary can be zlib compressed (or numpress'd) first. So to decode it, first you decode the base64 to binary, then if the cvParams say it's compressed, you decompress it, then you cast it to the appropriate binary array type according to the data type cvParam. For example, in C/C++, a little endian array of 64-bit float bytes can be casted directly to `double*`.

wkumler, 2020-11-02:
Fantastic, thank you! That's all I needed to know.

closed 2020-11-02 by wkumler